Ultimaker/Cura #382: 15.09.81: Lift head issue?

CCS86, 2015-09-11:
I don't have this one hammered down, but twice, as I went setting by setting making changes, and watching it successfully reslice, after checking "lift head", it never attempted to reslice. Once, I clicked "lift head" and it sliced, then I changed full fan at layer to 3, and no reslice.

Can someone else test this?

nallath, 2015-09-17:
I've not been able to reproduce this. In theory every setting change should trigger a reslice.

CCS86, 2015-09-17:
Yeah, I couldn't get it nailed down to a specific order of steps to reproduce. But, multiple times after starting with a default profile, and working downwards changing settings, observing a reslice after each change, some combination of changing the minimum layer time and enabling cool lift would no longer reslice. 

I'll wait for .82 and try again.

nallath, 2015-09-18:
It could be that it sees one of the values as invalid (but not min / max error) and refuse to reslice. We didn't notify the user of this, which is changed in the new version.

The log file should have a warning about it though, so you could check if this happened.

CCS86, 2015-09-21:
Looking good so far in .83

closed 2015-09-23 by nallath