Ultimaker/Cura #383: Empty context menus

Ghostkeeper, 2015-09-11:
Version: 15.09.81
Platform: Ubuntu 15.04 (Unity)
Qt version: 5.4.1
PyQt version: 5.4.1

Steps to reproduce:

* Right-click the 3D view of Cura a couple of times, or a setting, or the save button.

Expected behaviour:

* A context menu appears at the cursor, with options applying to whatever you right-clicked on.

Observed behaviour:

* Sometimes, an empty context menu pops up at the cursor. The menu is of the correct size, but is completely white.
* The context menu appears empty about ⅓ of the time.
* Clicking or hovering the context menu has no effect. No highlighted bar appears when hovering. Clicking the context menu makes the menu disappear, but does nothing else.

I can't make a screenshot of this as my screenshot tool doesn't seem to work while a context menu is open, so I hope you'll be able to go by my description.

nallath, 2015-09-23:
I haven't been able to reproduce this in any way. I'm guessing this is either an upstream bug or something with your configuration. Do you still have this issue?

Ghostkeeper, 2015-09-24:
The issue persists in version fdd485a884daaaacb170471d502048fc9b6c0d83.

TMHogenhout, 2015-10-06:
I saw it in testing as well. It was reproduced in every test-setup by clicking slightly left from the arrow in the dropdown menu. 

2015-10-01 14:55:09,374 - DEBUG - Backend connected on port 49674
2015-10-01 14:55:09,374 - DEBUG - Setting active view to MeshView
2015-10-01 14:55:10,074 - DEBUG - Loaded theme /usr/share/cura/resources/themes/cura
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/Menu.qml:146: TypeError: Cannot read property '__contentItem' of null
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/Menu.qml:146: TypeError: Cannot read property '__contentItem' of null
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/Menu.qml:146: TypeError: Cannot read property '__contentItem' of null
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/Menu.qml:146: TypeError: Cannot read property '__contentItem' of null
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/Menu.qml:146: TypeError: Cannot read property '__contentItem' of null
2015-10-01 14:55:10,413 - DEBUG - Preparing to send slice data to engine.

Ghostkeeper, 2015-10-07:
I also see those errors in the console when I launch any Uranium application. But the error doesn't only show up when clicking left of the arrow in a dropdown menu. It also happens when clicking right on the arrow as well as when right-clicking the 3D view.

BagelOrb, 2016-04-26:
No problem anymore

nallath, 2016-04-28:
This is a unity issue right?

Ghostkeeper, 2016-04-28:
Still occurs: [Photo, since Unity can't take screenshots while a context menu is open.](http://dulek.net/work/EmptyContextMenu.jpg)

It's only been observed with Unity.

awhiemstra, 2016-04-28:
That is not a context menu. ;)

In any case, this is an upstream issue, either in Ubuntu or in Qt. Since it only happens on Ubuntu with Unity I am inclined to blame Ubuntu/Unity, but I recall seeing a bug report about it in the Qt bug tracker.

Ghostkeeper, 2016-04-29:
It's not a context menu but the same sort of widget. The same thing happens with actual context menus ;)

BagelOrb, 2017-07-21:
I'm voting to close this even if it's still reproducible. It doesn't reproduce on Windows or Linux Mint with me so this issue seems very specific to your OS.

We should close this issue because there's 23 pages of issues in Cura!

closed 2017-07-21 by BagelOrb