Ultimaker/Cura #384: Debugging for plugins in 15.06.03?

mpbomil, 2015-09-11:
On macs it's easy to find the debug mode by showing package contents and running "cura', but what about windows?

nallath, 2015-09-11:
I don't quite understand what you mean by this. There is no debug mode for Cura (aside from running it from source). 
Windows simply the files that are in the package into the install folder.

peteruithoven, 2015-09-11:
I think the question is whether there is an easy way to see custom logging / prints, preferably without compiling Cura from source. Logs created with something like Logger.debug(). Something like a console window. 
I think I see these on On OS X when I start Cura through the commandline using 
``` bash
$ .../Cura.app/Contents/MacOS/Cura
```

mpbomil, 2015-09-11:
I found what I was looking for in: https://github.com/Ultimaker/Cura#logging-issues
Thanks!

closed 2015-09-11 by mpbomil

nallath, 2015-09-11:
Oh, custom logging that are created through Logger.log('d',"message") are shown in the console. They are also written to the log file.