Ultimaker/Cura #385: [master] [nvidia 355] shader compilation failed

thopiekar, 2015-09-11:
Don't know which commit causes this problem, because I didn't run Cura daily but this is what I get at the moment:

```
QOpenGLShaderProgram::uniformLocation( dpr ): shader program is not linked
QOpenGLShaderProgram: could not create shader program
QOpenGLShader: could not create shader
QOpenGLShader: could not create shader
shader compilation failed: 
""
QOpenGLShaderProgram::uniformLocation( matrix ): shader program is not linked
QOpenGLShaderProgram::uniformLocation( color ): shader program is not linked
QOpenGLShaderProgram::uniformLocation( textureScale ): shader program is not linked
QOpenGLShaderProgram::uniformLocation( dpr ): shader program is not linked
QOpenGLShaderProgram: could not create shader program
QOpenGLShader: could not create shader
QOpenGLShader: could not create shader
shader compilation failed: 
""
QOpenGLShaderProgram::uniformLocation( qt_Matrix ): shader program is not linked
QOpenGLShaderProgram: could not create shader program
QOpenGLShader: could not create shader
QOpenGLShader: could not create shader
shader compilation failed: 
""
QOpenGLShaderProgram::uniformLocation( matrix ): shader program is not linked
QOpenGLShaderProgram::uniformLocation( color ): shader program is not linked
QOpenGLShaderProgram::uniformLocation( textureScale ): shader program is not linked
QOpenGLShaderProgram::uniformLocation( dpr ): shader program is not linked
QOpenGLShaderProgram: could not create shader program
QOpenGLShader: could not create shader
QOpenGLShader: could not create shader
shader compilation failed: 
""
QOpenGLShaderProgram::uniformLocation( matrix ): shader program is not linked
QOpenGLShaderProgram::uniformLocation( color ): shader program is not linked
QOpenGLShaderProgram::uniformLocation( textureScale ): shader program is not linked
QOpenGLShaderProgram::uniformLocation( dpr ): shader program is not linked
QOpenGLShaderProgram: could not create shader program
QOpenGLShader: could not create shader
QOpenGLShader: could not create shader
shader compilation failed: 
""
QOpenGLShaderProgram::uniformLocation( matrix ): shader program is not linked
QOpenGLShaderProgram::uniformLocation( color ): shader program is not linked
QOpenGLShaderProgram::uniformLocation( textureScale ): shader program is not linked
QOpenGLShaderProgram::uniformLocation( dpr ): shader program is not linked
QOpenGLShaderProgram: could not create shader program
QOpenGLShader: could not create shader
QQuickCustomMaterialShader: Shader compilation failed:
""
QOpenGLShader: could not create shader
QOpenGLShader: could not create shader
QOpenGLShaderProgram::uniformLocation( qt_Matrix ): shader program is not linked
QOpenGLShaderProgram::uniformLocation( src ): shader program is not linked
QOpenGLShaderProgram::uniformLocation( qt_SubRect_src ): shader program is not linked
QOpenGLShaderProgram::uniformLocation( color ): shader program is not linked
QOpenGLShaderProgram::uniformLocation( qt_Opacity ): shader program is not linked
Speicherzugriffsfehler (Speicherabzug geschrieben) - aka seg. fault
thopiekar@home:~$ dmesg
[   30.278707] nvidia: module verification failed: signature and/or required key missing - tainting kernel
[   30.284426] NVLINK: Nvlink Core is being initialized, major device number 246
[   30.284555] [drm] Initialized nvidia-drm 0.0.0 20150116 for 0000:02:00.0 on minor 0
[   30.284572] NVRM: loading NVIDIA UNIX x86_64 Kernel Module  355.06  Tue Jul 28 21:26:50 PDT 2015
[ 8344.185779] cura_app.py[3275]: segfault at 40 ip 00007f2b860f3385 sp 00007ffe12ec5220 error 4 in libnvidia-glcore.so.355.06[7f2b85032000+126e000]
[ 8376.455519] cura_app.py[3980]: segfault at 40 ip 00007fdb12343385 sp 00007fffd2f0d0c0 error 4 in libnvidia-glcore.so.355.06[7fdb11282000+126e000]
```

thopiekar, 2015-09-11:
Found a possible solution: #131

thopiekar, 2015-09-11:
Duplicate of #88, which is closed for some reason :disappointed:

nallath, 2015-09-11:
In that case installing the right package solved it.

thopiekar, 2015-09-11:
Ok, will give it a try.

thopiekar, 2015-09-11:
Installed python3-opengl, but no changes.

thopiekar, 2015-09-11:
Found the reason at launchpad: https://bugs.launchpad.net/ubuntu/+source/python-qt4/+bug/941826
Will add a pull request soon based on #131

thopiekar, 2015-09-11:
My PR https://github.com/Ultimaker/Cura/pull/387 based on #131 fixes the problem + installing ```python3-opengl```

nallath, 2015-09-11:
Your PR has been merged, so this issue can be closed.

closed 2015-09-11 by nallath