Ultimaker/Cura #386: Custom support (like simplify3d, craftware and meshmixer)

Patola, 2015-09-11:
Since Cura is undergoing such a huge change, isn't it the best time to provide customized support placement like in Simplify3D, CraftUnique Craftware and Autodesk Meshmixer? That would be the silver bullet that would put Cura ahead of all the other slicers and print hosts.

The way this works in Simplify3D is: you have a dialog box which lets you add support towers using the cursor, remove support towers using the cursor, run the automatic support algorithm to generate the support towers in place according to the parameters set in the configuration and clear all supports in the current part. It colors the supports darker than the actual part so it's easy to distinguish.

If you have customized supports in place but the support option is off, it warns you if you really want this off, lets you select yes or no, then proceeds.

nallath, 2015-09-11:
I fully agree. This feature has been on our wish list for quite some time. I'm going to push for this feature to be included in the next (16.01) release.

Ghostkeeper, 2015-09-11:
I agree that it is important for Cura to implement this quickly (though the upcoming release is already in feature freeze right now).

Outline of the most challenging steps needed to accomplish this:

* Have CuraEngine split support into segments that can be enabled/disabled via the command line options, or else define 3-dimensional regions (or 2-dimensional if only support touching buildplate is generated?) in which support is enabled/disabled.
* Render the generated support in Cura, and not just in layer view.
* Compute cursor collision regions to allow the user to interact with support, or make a tool that creates polygon regions where support is generated/disabled (depending on the choice made in the first step).
* Send the allowed/disallowed regions or pieces to CuraEngine via the command line options.

Feel free to add anything to that list. I'm also hesitant about using the command line to communicate regions on the model. Perhaps that can be communicated via the mesh?

nickthetait, 2015-09-11:
*drools*

How about using a separate mesh file just to indicate where support struts are needed?

CCS86, 2015-09-12:
Glad to hear it could be in the works soon! I've been wanting this badly.

Actually, when I first saw the "towers" support in the new release, that's what I thought it was.

nallath, 2015-09-14:
There are tons of way we could do this and we haven't made a decision on what is the best way to do it. A custom mesh should already be possible for the new version btw. Just add two meshes, merge them and give the 'support mesh' custom settings.

BagelOrb, 2017-07-21:
There's a really user-unfriendly way to do this now, using the per-object settings Support Mesh or Anti-Overhang Mesh....

nickthetait, 2017-07-23:
Much better than nothing :sunglasses:

Ghostkeeper, 2017-08-07:
Well, not *much* better, if there are so few people that get to use this...

quasipedia, 2017-11-10:
Any news on how this is coming together (if at all?). :)

fieldOfView, 2017-11-10:
The unfriendly way may be getting a little less unfriendly:
https://github.com/Ultimaker/Cura/pull/2758

Vandrasc, 2018-04-09:
Support blocker has been implemented, closed.

quasipedia, 2018-04-10:
Support blocker is _not_ custom support.  It's a quick hack to give some extra control over support material.  This issue should be left open (or closed as "won't fix").

BagelOrb, 2018-04-10:
I agree. A full custom support system is not going to be implemented soon, though.

ChrisTerBeke, 2018-04-10:
Yay, another infinite open ticket....

ianpaschal, 2018-04-10:
Also because this would really require totally reworking how support functions and need lots more UI and maybe even some big changes to the workflow (i.e. a step between loading a model, tweaking settings, and slicing) I think it's fair to categorize it as a new feature vs. an improvement.

diegopradogesto, 2018-07-05:
You can checkout a new plugin in the Toolbox called "Custom Supports". It uses the same technique as the Support Blocker, but the other way around :-). Try it out!

Appesteijn, 2018-11-22:
We might have somethin in the pipeline, when it is done your the first to know :)

closed 2018-11-22 by Appesteijn