Ultimaker/Cura #389: uncaught exception

inventinside, 2015-09-13:
Starting the app after 2 days of use.
Now it throws this every time I start the app.

Version: 15.06.03
Platform: Windows-8-6.2.9200
Qt: 5.4.1
PyQt: 5.4.1

Exception:
Traceback (most recent call last):
  File "cura_app.py", line 9, in <module>
  File "c:\dev\uranium\UM\Application.py", line 329, in getInstance
  File "c:\dev\cura\cura\CuraApplication.py", line 56, in __init__
  File "c:\dev\uranium\UM\Qt\QtApplication.py", line 76, in __init__
  File "c:\dev\uranium\UM\Application.py", line 355, in loadMachines
  File "c:\dev\uranium\UM\Settings\MachineSettings.py", line 87, in loadValuesFromFile
  File "c:\dev\python34\lib\configparser.py", line 672, in read
  File "c:\dev\python34\lib\configparser.py", line 1044, in _read
configparser.DuplicateSectionError: While reading from 'C:\\Users\\povil/AppData/Local/cura\\settings\\Ultimaker+2+Extended.cfg' [line 38]: section 'blackmagic' already exists

Ghostkeeper, 2015-09-13:
Probably related to that bug where the settings files get corrupt by the Python file writer, with race conditions (#346) or general corruption (#252).

awhiemstra, 2015-09-13:
Yeah, it is similar. Please remove the mentioned file (C:\Users\povil/AppData/Local/cura\settings\Ultimaker+2+Extended.cfg) and try again. In the meantime I will look into doing more atomic file writes.

nallath, 2015-09-15:
Duplicate; were working on fixing this.

closed 2015-09-15 by nallath