Ultimaker/Cura #391: you delete a object but it still prints!

robomaniac, 2015-09-13:
Hello

I drag and drop new .STL file in Cura 15.06.03 (The one on the RIGHT)
[image]

I right click on the left object and click "delete". (I deleted the LEFT part)
[image]

I then click "save to SD card"

I go print it and print both parts instead of the single one!!!

Ghostkeeper, 2015-09-13:
Maybe it fails to trigger a re-slice after deleting it?

Xisle, 2015-09-13:
Maybe it's already fixed in 15.09.8x?

robomaniac, 2015-09-13:
I have the new 15.09.81 and will let you guys know.
That bug already happen at least 3 times.

I will take great note when I try to reproduce this bug.

Thank you

nallath, 2015-09-16:
Should be fixed now (starting a new print now clears existing data)

closed 2015-09-16 by nallath