Ultimaker/Cura #392: Mouse cursor is offset when moving object

robomaniac, 2015-09-13:
In Cura 15.06.03 or any other version of cura (at least the last 2)
Every time I move a object, the object shift randomly around.
Since I can not printscreen a mouse cursor, I placed one in paint so you know where it is in that demo.

It does the same thing if I click on the green or red arrow.
[image]

So you understand. I open the STL file.
I then click on of one of the 4 holes and I start dragging the part.
Then my part shift randomly (I think it's random. I did not see any pattern when I move parts in cura since day one)
then I can move it around but it's offset.

The mouse cursor does not move, it's the part underneath that moves.

fieldOfView, 2015-09-13:
This had already been fixed in the 15.10 open beta. Check out the forum for a link

robomaniac, 2015-09-13:
Thank you! I have installed 15.10 and will test that one instead. Lots of new stuff!
https://ultimaker.com/en/community/view/17295-cura-1510-open-beta

Ghostkeeper, 2015-09-13:
Duplicate of #100.

closed 2015-09-13 by Ghostkeeper