Ultimaker/Cura #394: 15.09.81: Material settings not in right pane.

robomaniac, 2015-09-14:
I do not see any of these settings in the right pane...

[image]

nallath, 2015-09-14:
Did you try this with 15.9.81?

robomaniac, 2015-09-14:
yes that is 15.9.81

nallath, 2015-09-14:
I can't reproduce this. It works perfectly okay on my system. Could you post the log files?

robomaniac, 2015-09-14:
Ok, tonight I will search and post the log file for both #395 and this one.

When I have the log file, I simply reply to this thread following these guidelines?
https://github.com/Ultimaker/Uranium/wiki/Bug-Reporting-Template

thanks

nallath, 2015-09-14:
We prefer it if you do that for all bug reports. It saves us a lot of time asking for the standard stuff.

robomaniac, 2015-09-15:
Application Version: 15.09.81
Platform: Windows 7 SP1
Qt: 5.4.1
PyQt: 5.4.1
Display Driver: Quadro K4000M (ver. 9.18.13.4105)

Let me know if do something wrong. (This is not the entire cura.log file)
2015-09-12 17:37:07,179 - INFO - Loaded plugin FileLogger
2015-09-12 17:37:07,399 - INFO - Loaded plugin LocalFileStorage
2015-09-12 17:37:07,409 - INFO - Loaded plugin LayerView
2015-09-12 17:37:07,423 - INFO - Loaded plugin MeshView
2015-09-12 17:37:07,438 - INFO - Loaded plugin WireframeView
2015-09-12 17:37:07,444 - INFO - Loaded plugin OBJReader
2015-09-12 17:37:07,448 - INFO - Loaded plugin STLReader
2015-09-12 17:37:07,455 - INFO - Loaded plugin STLWriter
2015-09-12 17:37:07,460 - INFO - Loaded plugin GCodeWriter
2015-09-12 17:37:07,472 - INFO - Loaded plugin CameraTool
2015-09-12 17:37:07,482 - INFO - Loaded plugin MirrorTool
2015-09-12 17:37:07,500 - INFO - Loaded plugin RotateTool
2015-09-12 17:37:07,511 - INFO - Loaded plugin ScaleTool
2015-09-12 17:37:07,522 - INFO - Loaded plugin SelectionTool
2015-09-12 17:37:07,534 - INFO - Loaded plugin TranslateTool
2015-09-12 17:37:07,549 - INFO - Checking for new version of cura
2015-09-12 17:37:07,549 - INFO - Loaded plugin UpdateChecker
2015-09-12 17:37:07,578 - INFO - Loaded plugin USBPrinting
2015-09-12 17:37:07,580 - INFO - Loaded plugin CuraEngineBackend
2015-09-12 17:37:08,677 - DEBUG - Loaded theme C:/Program Files (x86)/Cura_15.06.03/resources/themes/cura
2015-09-12 17:37:09,111 - DEBUG - Loaded a mesh with 1188 vertices
2015-09-12 17:37:09,367 - INFO - Started engine process: C:\Program Files (x86)\Cura_15.06.03\CuraEngine.exe
2015-09-12 17:37:09,563 - DEBUG - Backend connected on port 49674
2015-09-12 17:37:10,150 - DEBUG - Killing engine process
2015-09-12 17:37:10,152 - INFO - Backend crashed or closed. Restarting...
2015-09-12 17:37:10,421 - INFO - Started engine process: C:\Program Files (x86)\Cura_15.06.03\CuraEngine.exe
2015-09-12 17:37:10,451 - DEBUG - Backend connected on port 49674
2015-09-12 17:37:16,606 - INFO - Loaded plugin FileLogger
2015-09-12 17:37:16,802 - INFO - Loaded plugin LocalFileStorage
2015-09-12 17:37:16,812 - INFO - Loaded plugin LayerView
2015-09-12 17:37:16,827 - INFO - Loaded plugin MeshView
2015-09-12 17:37:16,842 - INFO - Loaded plugin WireframeView
2015-09-12 17:37:16,848 - INFO - Loaded plugin OBJReader
2015-09-12 17:37:16,851 - INFO - Loaded plugin STLReader
2015-09-12 17:37:16,858 - INFO - Loaded plugin STLWriter
2015-09-12 17:37:16,864 - INFO - Loaded plugin GCodeWriter
2015-09-12 17:37:16,875 - INFO - Loaded plugin CameraTool
2015-09-12 17:37:16,884 - INFO - Loaded plugin MirrorTool
2015-09-12 17:37:16,903 - INFO - Loaded plugin RotateTool
2015-09-12 17:37:16,915 - INFO - Loaded plugin ScaleTool
2015-09-12 17:37:16,926 - INFO - Loaded plugin SelectionTool
2015-09-12 17:37:16,938 - INFO - Loaded plugin TranslateTool
2015-09-12 17:37:16,954 - INFO - Checking for new version of cura
2015-09-12 17:37:16,954 - INFO - Loaded plugin UpdateChecker
2015-09-12 17:37:16,972 - INFO - Loaded plugin USBPrinting
2015-09-12 17:37:16,975 - INFO - Loaded plugin CuraEngineBackend
2015-09-12 17:37:17,963 - DEBUG - Loaded theme C:/Program Files (x86)/Cura_15.06.03/resources/themes/cura
2015-09-12 17:37:18,397 - DEBUG - Loaded a mesh with 636 vertices
2015-09-12 17:37:18,416 - INFO - Started engine process: C:\Program Files (x86)\Cura_15.06.03\CuraEngine.exe
2015-09-12 17:37:18,562 - DEBUG - Backend connected on port 49674
2015-09-12 17:37:19,648 - DEBUG - Killing engine process
2015-09-12 17:37:19,650 - INFO - Backend crashed or closed. Restarting...
2015-09-12 17:37:19,919 - INFO - Started engine process: C:\Program Files (x86)\Cura_15.06.03\CuraEngine.exe
2015-09-12 17:37:19,947 - DEBUG - Backend connected on port 49674
2015-09-12 17:37:37,612 - DEBUG - Loaded a mesh with 936 vertices
2015-09-12 17:37:43,449 - DEBUG - Killing engine process
2015-09-12 17:37:43,466 - INFO - Backend crashed or closed. Restarting...
2015-09-12 17:37:43,736 - INFO - Started engine process: C:\Program Files (x86)\Cura_15.06.03\CuraEngine.exe
2015-09-12 17:37:43,767 - DEBUG - Backend connected on port 49674
2015-09-12 17:37:46,313 - DEBUG - Killing engine process
2015-09-12 17:37:46,335 - INFO - Backend crashed or closed. Restarting...
2015-09-12 17:37:46,627 - INFO - Started engine process: C:\Program Files (x86)\Cura_15.06.03\CuraEngine.exe
2015-09-12 17:37:46,677 - DEBUG - Backend connected on port 49674
[... 664 lines cut ...]
2015-09-14 23:43:07,967 - INFO - Loaded plugin LocalFileOutputDevice
2015-09-14 23:43:07,988 - INFO - Loaded plugin PostProcessingPlugin
2015-09-14 23:43:08,016 - INFO - Loaded plugin RemovableDriveOutputDevice
2015-09-14 23:43:08,069 - INFO - Loaded plugin SliceInfoPlugin
2015-09-14 23:43:08,095 - INFO - Loaded plugin CameraTool
2015-09-14 23:43:08,128 - INFO - Loaded plugin MirrorTool
2015-09-14 23:43:08,159 - INFO - Loaded plugin PerObjectSettingsTool
2015-09-14 23:43:08,200 - INFO - Loaded plugin RotateTool
2015-09-14 23:43:08,234 - INFO - Loaded plugin ScaleTool
2015-09-14 23:43:08,268 - INFO - Loaded plugin SelectionTool
2015-09-14 23:43:08,304 - INFO - Loaded plugin TranslateTool
2015-09-14 23:43:08,341 - INFO - Checking for new version of cura
2015-09-14 23:43:08,341 - INFO - Loaded plugin UpdateChecker
2015-09-14 23:43:08,392 - INFO - Loaded plugin USBPrinting
2015-09-14 23:43:08,437 - INFO - Loaded plugin MeshView
2015-09-14 23:43:08,482 - INFO - Loaded plugin WireframeView
[... cut ...]
2015-09-14 23:43:08,884 - DEBUG - Setting value of setting layer_height to 0.2
2015-09-14 23:43:08,884 - DEBUG - Setting value of setting speed_print to 50
2015-09-14 23:43:08,885 - DEBUG - Setting value of setting layer_height to 0.06
2015-09-14 23:43:08,886 - DEBUG - Setting value of setting layer_height to 0.1
2015-09-14 23:43:08,888 - DEBUG - Backend connected on port 49674
2015-09-14 23:43:08,898 - DEBUG - Setting active view to MeshView
2015-09-14 23:43:10,738 - DEBUG - Loaded theme C:\Program Files (x86)\Cura_15.09.81\resources\themes\cura

nallath, 2015-09-15:
I don't see anything strange in the log files. Odd.

awhiemstra, 2015-09-15:
The issue here is a discrepancy between setting visibility and enabled settings. Since you are using an Ultimaker 2 we disable the material temperature settings as they are supposed to be set on the machine. The setting visibility screen does not account for enabled/disabled settings currently as it is intended to show whether or not the settings should be visible when they are enabled. That said, we might be able to add some kind of indication that this is happening.

robomaniac, 2015-09-15:
" *they are supposed to be set on the machine*"

I want a complete control over my machine. How can I disable that? it does not matter what I change in cura since they are overwrite in my printer. (it seems that I can not :disappointed: )

Please have a look at https://github.com/Ultimaker/Cura/issues/399
How can I avoid bad print if I don't have access to those settings? I think my bed temp is not high enough. This is ABS, I need ~90-100C

sure I can "tune" my printer while it prints (which is awesome) but I don't want to do that everytime.

ps - before I forget, I will uninstall and reinstall Cura see if that helps!

Thank you for all you work!

nallath, 2015-09-15:
Uninstalling or reinstalling won't have any effect on this; this has to do with how we designed it to be.

You -can- change those settings (eg; the slicer will accept it if you do and add those to the g-code). The printer will notify you if you change the temperature (It will say something like; this will override material settings). So Cura is leading in this regard!

If you want these features to be enabled, you're going to need to change the machine definition. These are found in the /resources/machines/ultimaker2.json There you can enable the material_print_temperature setting. I'm not a 100% sure if that's all that you need to do, but that should do the trick.

The official way of doing it is by adjusting the temperature on the machine itself. You can change this on the go by using the tune menu or you can change the material profiles on the machine.

robomaniac, 2015-09-15:
ok, I am not happy. Before getting upset, I will try to modifiy the **ultimaker2.json** file tonight.

I will find the file and change these values (this is copy past from github):
   "overrides": {
        "material_print_temperature": { "**enabled**": "False" },
        "material_bed_temperature": { "**enabled**": "False" },
        "material_diameter": { "**enabled**": "False" },
        "material_flow": { "**enabled**": "False" }
    }
I just modify the .json file and save. I don't need to compile anything, right?

Maybe it's different on my computer. If you look at my picture that could explain why they are **NOT** active in the right pane. I will **disabled** them and restart cura.

`The official way of doing it is by adjusting the temperature on the machine itself. You can change this on the go by using the tune menu or you can change the material profiles on the machine.`
I design by doing lots of iteration so it is long and frustrating to change the settings on the printer everytime! Exceptionally when I know the specific values to get a good print with my specific material.

Can you add the label **enhancement**? What I want, is a check box (or whatever) that will let me have full control over the settings of the machine. Like makerbot, I choose everything in the advanced settings. I tought the advanced tab would do that in Cura. 

I don't want the machine to override anything.

Do you know how much time I took tweeking those values to get the best print quality to then realise after all this, that the machine actually overides my values!!!!!! What a waste of time!

Can't wait to try this out tonight!
Merci!

robomaniac, 2015-09-15:
Ok, I spend a good amount of time to try to get this working. It does not work.

I have been playing with these line of code.

"overrides": {
        "material_print_temperature": { "enabled": "False" },
        "material_bed_temperature": { "enabled": "False" },
        "material_diameter": { "enabled": "False" },
        "material_flow": { "enabled": "False" }
    }

Here are the parameter I tried.
        "material_print_temperature": { "disabled": "False" },
        "material_print_temperature": { "disabled": "true" },
        "material_print_temperature": { "disable": "true" },
        "material_print_temperature": { "enabled": "true" },
        "material_print_temperature": { "enable": "true" },
        "material_print_temperature": { "default": 222.0 },

Then added a new printer, like the ultimaker 2 extended or the prusa i3. I was able to see the 
print temperaure, bed temperature, I was super happy, I then prepared my part, click export and it crash.
https://github.com/Ultimaker/Cura/issues/400

nallath, 2015-09-16:
You can also tune the machine (on the go) or manually add the M104 S[TEMPERATURE] command.

The stuff you're trying to do is actually more behaviour that we see from people who own an UMO (as thats more of a tinkerer machine). You're the first person in time that the um2 is out to say that this is annoying behaviour.

robomaniac, 2015-09-16:
ok, I will ask a question in a other way.

I bought a new filament, after tweaking, I realise that I get awesome print at 240C and a bed at 110C with a flow of 95%.
The default of ABS is 250(or 260) and bed at 90.
The PLA is lower ~230 and bed at 60.
So I can't use the default.

How do I print my material then? 
Yes, I can TUNE the machine but it's not logical to do that for every single print, I have a 750g of spool. What happen is this is my ultimate material that I love and want to print the rest of my life with :laughing: 

Regarding this **M104 S[TEMPERATURE] command.** that is gcode right?

nallath, 2015-09-16:
1. Materials - > Settings -> Customize 
2. Enter the settings that you want
3. In same menu press ; store as preset 

The m104 is a g-code command yeah.

robomaniac, 2015-09-17:
Fair enough!

I will have a post-it on my machine writting 
custom2 = alloy 910
custom3 = ClearScent ABS

I will also try the function "import settings from SD card"
I don't how to write the material settings to SD card. I don't see that option in Cura.

Thanks guys!

nallath, 2015-09-17:
Cura doesn't have that. The printer can write to the SD card by itself and you can then move it to another printer.

awhiemstra, 2015-09-25:
Closing this as everything seems to work as intended.

closed 2015-09-25 by awhiemstra