Ultimaker/Cura #397: 15.09.81: machine_platform_offset does not make any difference

nicanor-romero, 2015-09-14:
Changing this value in the machine settings has no effect in the mesh position.

In the Engine Log, this appears:

"Unrecognized data type in JSON: machine_platform_offset has type array"

nallath, 2015-09-14:
The engine log doesn't need the offset (but gets it anyway). That error can be safely ignored. For some reason the code that actually moves the platform mesh based on that setting has been commented out. 

If you feel like it, you could experiment a bit with the code that you can find in _onActiveMachineChanged of curaApplication.py. 

We're rather busy with fixing bigger issues, so this won't have a very high priority with us, sorry.

nicanor-romero, 2015-09-14:
Will do, thanks!

nallath, 2015-09-22:
Any progress with regards to this? I'm curious if you're able to get it working. We'd love to get more contributions from other people outside of UM.

awhiemstra, 2015-09-22:
Hmm, I was the one that commented out that code as it should be done by the Platform class, since that handles all other platform related settings. I just completely forgot to move the code there.

nallath, 2015-09-25:
Should be fixed in next release, we needed it as well.

closed 2015-09-25 by nallath