Ultimaker/Cura #399: 15.09.81: Default settings

robomaniac, 2015-09-15:
Hey guys, 

This could be a simple human error but I share it with you anyway.

------------------------------------------------------------------------
Application Version: 15.09.81
Platform: Windows 7 SP1
Qt: 5.4.1
PyQt: 5.4.1
Display Driver: Quadro K4000M (ver. 9.18.13.4105)

Steps to Reproduce: I was happy to try the new beta version. I started with my settings and when printing. Was not able to get it to print (see picture). I when back did some modification to Cura and got the same bad print. If you look at issue #394, I don't have the material setting in the right plane. So I could of had the bad bed temperature. **BUT** it does not matter what I change in cura since they are overwrite in my printer. (I do need to figure that one too) The printer set the bed to 90C and the nozzle to 260 for ABS. Even if I tell in Cura that I want Bed 100C and nozzle 250C, the printer does not do it.

Anyway, I end up using the old version to print. Even tonight I use the old version for all my prints.
That was my first experience with the beta version.

[image]

Here is some log file, you can read that it did set some value for print speed, etc.
------------------------------------------------------------------------
2015-09-14 18:27:36,025 - DEBUG - Loaded a mesh with 4800 vertices
2015-09-14 18:28:01,391 - DEBUG - Loaded a mesh with 4800 vertices
2015-09-14 18:28:43,973 - INFO - Loaded plugin FileLogger
2015-09-14 18:28:43,988 - INFO - Loaded plugin GCodeWriter
2015-09-14 18:28:44,005 - INFO - Loaded plugin LayerView
2015-09-14 18:28:44,025 - INFO - Loaded plugin LocalFileOutputDevice
2015-09-14 18:28:44,045 - INFO - Loaded plugin PostProcessingPlugin
2015-09-14 18:28:44,073 - INFO - Loaded plugin RemovableDriveOutputDevice
2015-09-14 18:28:44,140 - INFO - Loaded plugin SliceInfoPlugin
2015-09-14 18:28:44,165 - INFO - Loaded plugin CameraTool
2015-09-14 18:28:44,211 - INFO - Loaded plugin MirrorTool
2015-09-14 18:28:44,241 - INFO - Loaded plugin PerObjectSettingsTool
2015-09-14 18:28:44,283 - INFO - Loaded plugin RotateTool
2015-09-14 18:28:44,321 - INFO - Loaded plugin ScaleTool
2015-09-14 18:28:44,355 - INFO - Loaded plugin SelectionTool
2015-09-14 18:28:44,390 - INFO - Loaded plugin TranslateTool
2015-09-14 18:28:44,426 - INFO - Checking for new version of cura
2015-09-14 18:28:44,426 - INFO - Loaded plugin UpdateChecker
2015-09-14 18:28:44,477 - INFO - Loaded plugin USBPrinting
2015-09-14 18:28:44,518 - INFO - Loaded plugin MeshView
2015-09-14 18:28:44,560 - INFO - Loaded plugin WireframeView
2015-09-14 18:28:44,828 - INFO - Started engine process: C:\Program Files (x86)\Cura_15.09.81\CuraEngine.exe
2015-09-14 18:28:44,920 - DEBUG - Setting value of setting speed_print to 50
2015-09-14 18:28:44,920 - DEBUG - Setting value of setting layer_height to 0.2
2015-09-14 18:28:44,920 - DEBUG - Setting value of setting infill_sparse_density to 100
2015-09-14 18:28:44,921 - DEBUG - Setting value of setting layer_height to 0.06
2015-09-14 18:28:44,922 - DEBUG - Setting value of setting layer_height to 0.1
2015-09-14 18:28:44,924 - DEBUG - Backend connected on port 49674
2015-09-14 18:28:44,934 - DEBUG - Setting active view to MeshView
2015-09-14 18:28:46,912 - DEBUG - Loaded theme C:\Program Files (x86)\Cura_15.09.81\resources\themes\cura
2015-09-14 18:28:47,353 - DEBUG - Loaded a mesh with 4800 vertices
2015-09-14 18:28:54,803 - INFO - Loaded plugin FileLogger
2015-09-14 18:28:55,056 - INFO - Loaded plugin LocalFileStorage
2015-09-14 18:28:55,065 - INFO - Loaded plugin LayerView
2015-09-14 18:28:55,080 - INFO - Loaded plugin MeshView
2015-09-14 18:28:55,095 - INFO - Loaded plugin WireframeView
2015-09-14 18:28:55,101 - INFO - Loaded plugin OBJReader
2015-09-14 18:28:55,105 - INFO - Loaded plugin STLReader
2015-09-14 18:28:55,112 - INFO - Loaded plugin STLWriter
2015-09-14 18:28:55,118 - INFO - Loaded plugin GCodeWriter
2015-09-14 18:28:55,128 - INFO - Loaded plugin CameraTool
2015-09-14 18:28:55,138 - INFO - Loaded plugin MirrorTool
2015-09-14 18:28:55,157 - INFO - Loaded plugin RotateTool
2015-09-14 18:28:55,167 - INFO - Loaded plugin ScaleTool
2015-09-14 18:28:55,178 - INFO - Loaded plugin SelectionTool
2015-09-14 18:28:55,190 - INFO - Loaded plugin TranslateTool
2015-09-14 18:28:55,205 - INFO - Checking for new version of cura
2015-09-14 18:28:55,205 - INFO - Loaded plugin UpdateChecker
2015-09-14 18:28:55,237 - INFO - Loaded plugin USBPrinting
2015-09-14 18:28:55,239 - INFO - Loaded plugin CuraEngineBackend
[... 92 lines cut ...]
2015-09-14 23:43:07,967 - INFO - Loaded plugin LocalFileOutputDevice
2015-09-14 23:43:07,988 - INFO - Loaded plugin PostProcessingPlugin
2015-09-14 23:43:08,016 - INFO - Loaded plugin RemovableDriveOutputDevice
2015-09-14 23:43:08,069 - INFO - Loaded plugin SliceInfoPlugin
2015-09-14 23:43:08,095 - INFO - Loaded plugin CameraTool
2015-09-14 23:43:08,128 - INFO - Loaded plugin MirrorTool
2015-09-14 23:43:08,159 - INFO - Loaded plugin PerObjectSettingsTool
2015-09-14 23:43:08,200 - INFO - Loaded plugin RotateTool
2015-09-14 23:43:08,234 - INFO - Loaded plugin ScaleTool
2015-09-14 23:43:08,268 - INFO - Loaded plugin SelectionTool
2015-09-14 23:43:08,304 - INFO - Loaded plugin TranslateTool
2015-09-14 23:43:08,341 - INFO - Checking for new version of cura
2015-09-14 23:43:08,341 - INFO - Loaded plugin UpdateChecker
2015-09-14 23:43:08,392 - INFO - Loaded plugin USBPrinting
2015-09-14 23:43:08,437 - INFO - Loaded plugin MeshView
2015-09-14 23:43:08,482 - INFO - Loaded plugin WireframeView
2015-09-14 23:43:08,789 - INFO - Started engine process: C:\Program Files (x86)\Cura_15.09.81\CuraEngine.exe
2015-09-14 23:43:08,884 - DEBUG - Setting value of setting infill_sparse_density to 100
2015-09-14 23:43:08,884 - DEBUG - Setting value of setting layer_height to 0.2
2015-09-14 23:43:08,884 - DEBUG - Setting value of setting speed_print to 50
2015-09-14 23:43:08,885 - DEBUG - Setting value of setting layer_height to 0.06
2015-09-14 23:43:08,886 - DEBUG - Setting value of setting layer_height to 0.1
2015-09-14 23:43:08,888 - DEBUG - Backend connected on port 49674
2015-09-14 23:43:08,898 - DEBUG - Setting active view to MeshView
2015-09-14 23:43:10,738 - DEBUG - Loaded theme C:\Program Files (x86)\Cura_15.09.81\resources\themes\cura

robomaniac, 2015-09-15:
This week, I will try again the beta version with small parts. I just want to make sure it's not a human error.

nallath, 2015-09-15:
Thanks for looking into it!

nallath, 2015-09-23:
Any updates on this?

robomaniac, 2015-09-24:
I just tried again, I click save on SD card and it crash. So I saved on hardrive moved to SD and printed.
I had the default values... the fan was at 50% but the printer is set to ABS (fan to 0%) so that is not normal.

The part was printed but the quality was not that great (not like normal)

I am not using 15.09.81 anymore.  I am on 15.06.03. Is there something else I can do?
Enhancement are easy. I don't know what else I can try on 15.09.81 since I can't print normal.
thanks

nallath, 2015-09-24:
You could try 15.09.83. We fixed quite a few bugs since .81

robomaniac, 2015-09-27:
Is it here I should check for the latest one? I should check there often?
http://software.ultimaker.com/Cura_closed_beta/15.10/

I just downloaded 15.09.83, thanks for all your hard work.

Xisle, 2015-09-27:
Yes, check there. I have a page monitor plug-in watching for it to be updated.

nallath, 2015-09-28:
We have 3 month release cycles. We have a 'free for all' 2 month feature crunch and a 1 month bug fixing period (we're in the bug fixing now). During bug fixing we have a feature freeze. During the bug fixing we will quite often update the beta version.

BagelOrb, 2015-11-04:
I take it this bug has been solved.

Please inform me if I'm wrong.

closed 2015-11-04 by BagelOrb