Ultimaker/Cura #400: 15.09.81: Adding new printer and exporting = crash

robomaniac, 2015-09-15:
Application Version: 15.09.81
Platform: Windows 7-6.1.7601-SP1
Qt: 5.4.1
PyQt: 5.4.1
Display Driver: Quadro K4000M (ver. 9.18.13.4105)
-------------------------------------------------------------------------------------
I open Cura 15.09.81
In the right pane, I go next to Machine: and click the drop down menu, I select "add printer"
I select "ultimaker 2 extended"
I click finish
I drag my STL in Cura.
I insert SD card
I click "save to removable disk"
I click EJECT.
BOOM!!

I can not copy paste the error message, I was able to right click and saw "copy" but it did not work.
The frist try, I did ctrl+C and did not work.

so here is the print screen!

it's really the moment I click "eject" that the crash happen.

[image]

robomaniac, 2015-09-15:
15-09-15 17:55:50,189 - DEBUG - Setting value of setting material_print_temperature to 230
2015-09-15 18:02:17,973 - INFO - Loaded plugin FileLogger
2015-09-15 18:02:17,989 - INFO - Loaded plugin GCodeWriter
2015-09-15 18:02:18,006 - INFO - Loaded plugin LayerView
2015-09-15 18:02:18,026 - INFO - Loaded plugin LocalFileOutputDevice
2015-09-15 18:02:18,048 - INFO - Loaded plugin PostProcessingPlugin
2015-09-15 18:02:18,076 - INFO - Loaded plugin RemovableDriveOutputDevice
2015-09-15 18:02:18,128 - INFO - Loaded plugin SliceInfoPlugin
2015-09-15 18:02:18,155 - INFO - Loaded plugin CameraTool
2015-09-15 18:02:18,189 - INFO - Loaded plugin MirrorTool
2015-09-15 18:02:18,221 - INFO - Loaded plugin PerObjectSettingsTool
2015-09-15 18:02:18,262 - INFO - Loaded plugin RotateTool
2015-09-15 18:02:18,296 - INFO - Loaded plugin ScaleTool
2015-09-15 18:02:18,330 - INFO - Loaded plugin SelectionTool
2015-09-15 18:02:18,365 - INFO - Loaded plugin TranslateTool
2015-09-15 18:02:18,402 - INFO - Checking for new version of cura
2015-09-15 18:02:18,402 - INFO - Loaded plugin UpdateChecker
2015-09-15 18:02:18,406 - ERROR - Failed to check for new version. <urlopen error [Errno 11004] getaddrinfo failed>
2015-09-15 18:02:18,454 - INFO - Loaded plugin USBPrinting
2015-09-15 18:02:18,495 - INFO - Loaded plugin MeshView
2015-09-15 18:02:18,537 - INFO - Loaded plugin WireframeView
2015-09-15 18:02:18,576 - INFO - Started engine process: C:\Program Files (x86)\Cura_15.09.81\CuraEngine.exe
2015-09-15 18:02:18,669 - DEBUG - Setting value of setting layer_height to 0.2
2015-09-15 18:02:18,669 - WARNING - Tried to set value of non-user setting layer_height
2015-09-15 18:02:18,669 - DEBUG - Setting value of setting speed_print to 50
2015-09-15 18:02:18,669 - WARNING - Tried to set value of non-user setting speed_print
2015-09-15 18:02:18,669 - DEBUG - Setting value of setting infill_sparse_density to 100
2015-09-15 18:02:18,669 - WARNING - Tried to set value of non-user setting infill_sparse_density
2015-09-15 18:02:18,671 - DEBUG - Setting value of setting layer_height to 0.06
2015-09-15 18:02:18,671 - WARNING - Tried to set value of non-user setting layer_height
2015-09-15 18:02:18,671 - DEBUG - Setting value of setting layer_height to 0.1
2015-09-15 18:02:18,671 - WARNING - Tried to set value of non-user setting layer_height
2015-09-15 18:02:18,673 - DEBUG - Backend connected on port 49674
2015-09-15 18:02:18,683 - DEBUG - Setting active view to MeshView
2015-09-15 18:02:19,596 - DEBUG - Loaded theme C:\Program Files (x86)\Cura_15.09.81\resources\themes\cura
2015-09-15 18:02:37,136 - WARNING - No active machine found when trying to save setting visibility
2015-09-15 18:04:33,177 - INFO - Loaded plugin FileLogger
2015-09-15 18:04:33,192 - INFO - Loaded plugin GCodeWriter
2015-09-15 18:04:33,209 - INFO - Loaded plugin LayerView
2015-09-15 18:04:33,229 - INFO - Loaded plugin LocalFileOutputDevice
2015-09-15 18:04:33,250 - INFO - Loaded plugin PostProcessingPlugin
2015-09-15 18:04:33,277 - INFO - Loaded plugin RemovableDriveOutputDevice
2015-09-15 18:04:33,328 - INFO - Loaded plugin SliceInfoPlugin
2015-09-15 18:04:33,355 - INFO - Loaded plugin CameraTool
2015-09-15 18:04:33,388 - INFO - Loaded plugin MirrorTool
2015-09-15 18:04:33,419 - INFO - Loaded plugin PerObjectSettingsTool
2015-09-15 18:04:33,460 - INFO - Loaded plugin RotateTool
2015-09-15 18:04:33,494 - INFO - Loaded plugin ScaleTool
2015-09-15 18:04:33,528 - INFO - Loaded plugin SelectionTool
2015-09-15 18:04:33,563 - INFO - Loaded plugin TranslateTool
2015-09-15 18:04:33,599 - INFO - Checking for new version of cura
2015-09-15 18:04:33,600 - INFO - Loaded plugin UpdateChecker
2015-09-15 18:04:33,604 - ERROR - Failed to check for new version. <urlopen error [Errno 11004] getaddrinfo failed>
2015-09-15 18:04:33,650 - INFO - Loaded plugin USBPrinting
2015-09-15 18:04:33,691 - INFO - Loaded plugin MeshView
2015-09-15 18:04:33,733 - INFO - Loaded plugin WireframeView
2015-09-15 18:04:33,774 - INFO - Started engine process: C:\Program Files (x86)\Cura_15.09.81\CuraEngine.exe
2015-09-15 18:04:33,867 - DEBUG - Setting value of setting layer_height to 0.06
2015-09-15 18:04:33,867 - WARNING - Tried to set value of non-user setting layer_height
2015-09-15 18:04:33,867 - DEBUG - Setting value of setting layer_height to 0.1
2015-09-15 18:04:33,867 - WARNING - Tried to set value of non-user setting layer_height
2015-09-15 18:04:33,870 - DEBUG - Backend connected on port 49674
2015-09-15 18:04:33,879 - DEBUG - Setting active view to MeshView
2015-09-15 18:04:34,776 - DEBUG - Loaded theme C:\Program Files (x86)\Cura_15.09.81\resources\themes\cura
2015-09-15 18:06:12,106 - INFO - Loaded plugin FileLogger
2015-09-15 18:06:12,121 - INFO - Loaded plugin GCodeWriter
2015-09-15 18:06:12,139 - INFO - Loaded plugin LayerView
2015-09-15 18:06:12,159 - INFO - Loaded plugin LocalFileOutputDevice
2015-09-15 18:06:12,180 - INFO - Loaded plugin PostProcessingPlugin
2015-09-15 18:06:12,208 - INFO - Loaded plugin RemovableDriveOutputDevice
[... 258 lines cut ...]
2015-09-15 19:27:50,574 - INFO - Loaded plugin LayerView
2015-09-15 19:27:50,593 - INFO - Loaded plugin LocalFileOutputDevice
2015-09-15 19:27:50,614 - INFO - Loaded plugin PostProcessingPlugin
2015-09-15 19:27:50,641 - INFO - Loaded plugin RemovableDriveOutputDevice
2015-09-15 19:27:50,689 - INFO - Loaded plugin SliceInfoPlugin
2015-09-15 19:27:50,714 - INFO - Loaded plugin CameraTool
[... cut ...]
2015-09-15 19:27:51,223 - DEBUG - Setting value of setting layer_height to 0.06
2015-09-15 19:27:51,224 - DEBUG - Setting value of setting layer_height to 0.1
2015-09-15 19:27:51,226 - DEBUG - Backend connected on port 49674
2015-09-15 19:27:51,235 - DEBUG - Setting active view to MeshView
2015-09-15 19:27:53,060 - DEBUG - Loaded theme C:\Program Files (x86)\Cura_15.09.81\resources\themes\cura
2015-09-15 19:28:03,356 - DEBUG - Loaded a mesh with 3144 vertices
2015-09-15 19:28:07,427 - DEBUG - Writing to H:/PART_motor_mount_3.gcode

robomaniac, 2015-09-15:
ok, I did not add a new printer, I simply exported and it's really when I press EJECT that it crash!

oh la la!

[image]

robomaniac, 2015-09-15:
I have been playing with the ultimaker2.json file to get my other problem fix my problem with 
https://github.com/Ultimaker/Cura/issues/394
It is now the orignal file. That's the only thing I did tonight.

robomaniac, 2015-09-15:
ok, I unistall ultimaker and reinstall, same problem!!
this is the new printscreen but it's the same info has the other post.
[image]

robomaniac, 2015-09-16:
since the eject button was the problem, I save the file on computer, then copy it on SD card.

I was able to print my file BUT it was the default settings of the printer.
even if you can see the numbers I wrote in the .JSON file.
I wrote them in the ultimaker2.json and NOT the ultimakerextende.json

[image]

nallath, 2015-09-16:
This is a known issue. Not sure if we already fixed it, but it will be fixed for 15.10 release.

awhiemstra, 2015-09-25:
The eject crash should be fixed with the next build. If you still have other issues, please report them separately.

closed 2015-09-25 by awhiemstra