Ultimaker/Cura #401: 15.09.82 G Code generation for UMO

Xisle, 2015-09-18:
Application Version: 15.09.82
Platform: OS X 10.9.5

Steps to Reproduce:
Slice a Tri-force .stl model

Actual Results:
1) After homing the bed, a move to the center of the platform occurs before the part is printed.

2) Multiple passes are made over the same section for all of the bottom layers before beginning the hollow portion of the part.

3) 30 mm/s was chosen in Cura, but the G Code shows otherwise

4) There is at least one F270000 command in the G Code

5) At the end of the print, X/Y axes and the bed were homed.

Expected results:
None of the above.

Xisle, 2015-09-18:
[image]

Xisle, 2015-09-18:
https://github.com/Xisle/Cura/blob/master/tri-force_newCura.gcode

Xisle, 2015-09-18:
https://www.youtube.com/watch?v=vrrswd7CW7E

The first 3 seconds is all you need to watch

Xisle, 2015-09-18:
Just noticed the back right of the pyramid does not look right. It printed fine in 15.04.2.

[image]

Ghostkeeper, 2015-09-18:
UMO is obviously a newfag. And I keep being amazed by the amount of effort you put in bug reports. Thanks!

Xisle, 2015-09-21:
.83 appears to have fixed 1, 2 and 3.

4) I show many matches for "F270000" still in the .gcode

5) The end of the .gcode is a mess now...

[image]

awhiemstra, 2015-09-22:
The end GCode was already a mess, it is just a bit more of a mess now. ;)

One reason it looks so messy is that all the settings are exported, including machine settings like start/end GCode. This causes the duplicated start/end GCode at the end of the file.

mikenew12, 2015-09-24:
I'm on 15.09.83 and the multiple-passes issue is still very much a problem. When I go to slice this model:

[image]

parts of the bottom layers are doubled. You can see it in this visualization. Notice the top left, bottom left, and middle parts of the layer are darker because the lines have been drawn twice, however the right side of the layer is fine. Actually trying to print this causes the plastic to build up after a few layers and crash the print head. This is layer 4(ish):
[image]

The same thing happens towards the end of the print as it's printing top layers of the inset lettering (sorry that's hard to phrase correctly). This is around 10 layers before the end. Note that I turned infill to 0 just to make the visualization clearer; obviously this wouldn't actually print.
[image]

This is with almost completely default, "normal" settings. I adjusted the filament size, print temps, and start/end gcode, but other than that nothing special.

Here's the stl file: https://dl.dropboxusercontent.com/u/1667048/d6.stl

nallath, 2015-09-24:
Thats a nice visualisation. What tool do you use for that?

mikenew12, 2015-09-24:
@nallath https://github.com/hudbrog/gCodeViewer. This is with "emulate extrusion width" and "render lines slightly transparent" turned on.

mikenew12, 2015-09-30:
Just a quick update to this: as of 15.09.85 the multiple-passes issue is considerably worse. Parts of the print are being laid down at least 3 or 4 times (I can't tell for sure, because the visualization is just a black mass).

[image]

BagelOrb, 2015-10-01:
1) seems to already be solved.

2) There was a bug with duplicated skin parts for which I am quite ashamed. For every new skin part, all previous skin parts were reprinted.
It's solved in the next (pre-)release now.

Nice gcode viewer ^^

3) seems to already have been fixed.

4) Coasting is performed at a speed just under the speed with which it has just been printing. However, the percentage was taken to be a proportion, so the coasting speeds were 100 times too high. This has just been fixed.

5) The Cura profile string seems to be executed on an UMO. The M25 command is ignored by an UMO. After conversing with my colleagues, I've decided the current Cura profile string must be removed altogether. The frontend team will be working on reintroducing something similar to the profile string used to retrieve a profile from a gcode.

BagelOrb, 2015-10-01:
@Xisle What is the youtube video supposed to show? (https://www.youtube.com/watch?v=vrrswd7CW7E)

Are you trying to show the double skin parts bug (problem 2)?

If you can provide me with the tri-force model I will verify that it will be printed correctly.

closed 2015-10-01 by BagelOrb