Ultimaker/Cura #402: USB printing not working

GeneralNumpty, 2015-09-19:
When I try USB printing I get the temperature readings for the bed and nozzle but they dont increase. 

also If I wait for a while (20 mins with no apparent increase in the temps) and then just press print the printer will start to print but it doesn't re-home the head first and doesnt move the bed height. 

Thanks

nallath, 2015-09-19:
Could you post your log file? See https://github.com/Ultimaker/Cura#logging-issues for instructions.

GeneralNumpty, 2015-09-20:
I select the USB printing option
[image]
after a period of time 10 mins or so there is still no change in the temps (both head and bed), although the readings are approx correct when compared to what the printer is displaying so telemetary from printer to app seems to be working.

eventually I select the print button and the printer head immediately starts to move, but there is no attempt to feed the filament through just the head movement both X and Y. the bed doesnt move from home.

[image]

the printer front console cannot be accessed in this mode. once I cancel the print I can then access the printer console again.
*FYI*
when I start new Cura version with USB connected it seems to restart the printer, which pops up a screen saying printing with USB then to the normal config screen. is this normal behavior?
Printer is on Com8
had to screen print the engine log as I couldn't select it and copy before it became un-selected (seems to be refreshed to quickly).

[image]
*Log files*
**Cura.log**
2015-09-20 13:46:16,386 - INFO - Loaded plugin FileLogger
2015-09-20 13:46:16,618 - INFO - Loaded plugin LocalFileStorage
2015-09-20 13:46:16,629 - INFO - Loaded plugin LayerView
2015-09-20 13:46:16,647 - INFO - Loaded plugin MeshView
2015-09-20 13:46:16,666 - INFO - Loaded plugin WireframeView
2015-09-20 13:46:16,673 - INFO - Loaded plugin OBJReader
2015-09-20 13:46:16,677 - INFO - Loaded plugin STLReader
2015-09-20 13:46:16,686 - INFO - Loaded plugin STLWriter
2015-09-20 13:46:16,692 - INFO - Loaded plugin GCodeWriter
2015-09-20 13:46:16,706 - INFO - Loaded plugin CameraTool
2015-09-20 13:46:16,717 - INFO - Loaded plugin MirrorTool
2015-09-20 13:46:16,749 - INFO - Loaded plugin RotateTool
2015-09-20 13:46:16,762 - INFO - Loaded plugin ScaleTool
2015-09-20 13:46:16,775 - INFO - Loaded plugin SelectionTool
2015-09-20 13:46:16,788 - INFO - Loaded plugin TranslateTool
2015-09-20 13:46:16,806 - INFO - Checking for new version of cura
2015-09-20 13:46:16,806 - INFO - Loaded plugin UpdateChecker
2015-09-20 13:46:16,829 - INFO - Loaded plugin USBPrinting
2015-09-20 13:46:16,830 - DEBUG - Attempting to connect to COM8
2015-09-20 13:46:16,833 - INFO - Loaded plugin CuraEngineBackend
2015-09-20 13:46:17,464 - DEBUG - Starting baud rate detection...
2015-09-20 13:46:17,465 - DEBUG - Trying baud rate 250000
2015-09-20 13:46:18,101 - DEBUG - Loaded theme C:/Program Files (x86)/Cura_15.06.03/resources/themes/cura
2015-09-20 13:46:19,494 - INFO - Started engine process: C:\Program Files (x86)\Cura_15.06.03\CuraEngine.exe
2015-09-20 13:46:19,533 - INFO - Printer connection listen thread started for COM8
2015-09-20 13:46:19,533 - INFO - Established printer connection on port COM8
2015-09-20 13:46:19,719 - DEBUG - Backend connected on port 49674
2015-09-20 13:53:18,098 - DEBUG - Loaded a mesh with 432 vertices
2015-09-20 14:21:28,383 - INFO - Printer connection listen thread stopped for COM8

**DXdiag:**
------------------
System Information
------------------
Time of this report: 9/20/2015, 13:22:35
       Machine name: None
   Operating System: Windows 7 Home Premium 64-bit (6.1, Build 7601) Service Pack 1 (7601.win7sp1_gdr.150715-0600)
           Language: English (Regional Setting: English)
System Manufacturer: INTEL_
       System Model: DG33FB__
               BIOS: Default System BIOS
          Processor: Intel(R) Core(TM)2 Quad CPU    Q6600  @ 2.40GHz (4 CPUs), ~2.4GHz
             Memory: 6144MB RAM
Available OS Memory: 6074MB RAM
          Page File: 3026MB used, 9120MB available
        Windows Dir: C:\Windows
    DirectX Version: DirectX 11
DX Setup Parameters: Not found
   User DPI Setting: Using System DPI
 System DPI Setting: 96 DPI (100 percent)
    DWM DPI Scaling: Disabled
     DxDiag Version: 6.01.7601.17514 64bit Unicode

[... 609 lines cut ...]

---------------
EVR Power Information
---------------
Current Setting: {5C67A112-A4C9-483F-B4A7-1D473BECAFDC} (Quality) 
  Quality Flags: 2576
    Enabled:
    Force throttling
    Allow half deinterlace
    Allow scaling
    Decode Power Usage: 100
  Balanced Flags: 1424
    Enabled:
    Force throttling
    Allow batching
    Force half deinterlace
    Force scaling
    Decode Power Usage: 50
  PowerFlags: 1424
    Enabled:
    Force throttling
    Allow batching
    Force half deinterlace
    Force scaling
    Decode Power Usage: 0

nallath, 2015-09-20:
The connection seems to have been succesfull according to the logs. I will look into it more on monday.

nallath, 2015-09-23:
I see you're using 15.06. We're currently at the (beta) version of 15.09.83. Could you check if you have the same issues there? You can download it at http://software.ultimaker.com/Cura_closed_beta/15.10/

awhiemstra, 2015-09-25:
Looks like you are using an Ultimaker 2 but from the sound of it you are printing with an Ultimaker Original or similar? Or does the UM2 not properly execute its pre-print sequence when printing over USB?

thopiekar, 2016-01-13:
I can confirm this bug on Cura 2.1 (nightly build).
When I click on the print button the temperature is still decreasing and the progress bar is moving. I beleaved it is transferring the code to the RAM on the printer and then it will execute it, but after it has finished transferring the dialog has just closed.

nickthetait, 2016-01-14:
Are those nightly builds available somewhere?

thopiekar, 2016-01-14:
Yes, for Ubuntu at launchpad.
* Built using ```2.1``` branch: https://launchpad.net/~thopiekar/+archive/ubuntu/cura
* Built using ```master``` branch: https://launchpad.net/~thopiekar/+archive/ubuntu/cura-master

BagelOrb, 2017-07-21:
Issue is too old to provide info for fixing this if it even still is a problem

closed 2017-07-21 by BagelOrb