Ultimaker/Cura #403: 3D view becomes grey when opening STL file

MarijnHeeren, 2015-09-19:
Application Version: 15.06.03
Platform: OS X 10.10.5

Steps to Reproduce:
Load an STL. Right after it loads the 3D view becomes grey.

Actual Results:
3D view is grey. You can click it and see the controls to move an object. See below:
[image]

Expected results:
See the object, see the build platform.

nallath, 2015-09-19:
Could you try if it works with 15.09.82? You can find it at software.ultimaker.com/Cura_closed_beta/15.10/

nallath, 2015-09-23:
And the next version is also up. Could you check if it works there?

awhiemstra, 2015-09-23:
This is a duplicate of #110 that was already fixed.

closed 2015-09-23 by awhiemstra