Ultimaker/Cura #404: 15.09.82: Won't Launch

CCS86, 2015-09-19:
Hi guys,

Just installed .82 and found that it won't launch, and gives no error. Any ideas on how to troubleshoot this?

thopiekar, 2015-09-19:
OS? release? Were is it running on?

CCS86, 2015-09-19:
Sorry, hadn't had my coffee yet ;)

I am running Windows 10. Cura is installed in the default location on the primary drive.

nallath, 2015-09-19:
Post the logs.

CCS86, 2015-09-19:
Where are they stored again? I looked in ProgramData and AppData, and didn't find anything this time.

nallath, 2015-09-19:
https://github.com/Ultimaker/Cura#logging-issues

CCS86, 2015-09-19:
Cura has not created the AppData directory, but here is my dxdiag log.

(I gave it a jpg extension to push straight here. Just download and rename to .txt. Hopefully that works)

[image]

Ghostkeeper, 2015-09-19:
That log doesn't mention Cura, since Cura doesn't use DirectX. Ctrl+F shows up nothing.

thopiekar, 2015-09-19:
I know it is offtopic, but since you use an updater as plugin why is the version needed in the application path? (Sorry for starting this here - I'm just interested)

CCS86, 2015-09-19:
[image]

I know one of the other logs would be far more helpful, but apparently Cura isn't making it far enough to even create the Appdata directory.

.80 and .81 both launched fine on Windows 10 for me. This is a first. 

Yes, I have definitely launched Cura, multiple times. I tried all the different "compatibility modes", as administrator, with both of my video cards. Same. Just a brief hourglass, then nothing.

Is there a command line parameter to enable additional logging?

nallath, 2015-09-20:
Nope. If it doesn't even get to the logging, it dies fairly early into the process.

CCS86, 2015-09-21:
Fixed in .83!

awhiemstra, 2015-09-21:
Yeah there was an issue with fcntl on Windows (Ultimaker/Uranium#69) that caused the initial build for Windows to fail. I replaced the .82 build with a correct one but you apparently ended up with the incorrect build. Glad to hear it is all solved for .83.

closed 2015-09-21 by awhiemstra