Ultimaker/Cura #405: Infill density: Clearing the field crashes application

thopiekar, 2015-09-19:
When clearing the "Infill Density" field in Cura I get the following output. The message makes sense as there is nothing in the field, but it makes the App crash :disappointed: 

```
Version: 15.09.82
Platform: Linux-4.2.0-10-generic-x86_64-with-Ubuntu-15.10-wily
Qt: 5.4.2
PyQt: 5.4.2

Exception:
Traceback (most recent call last):
  File "/usr/lib/cura/plugins/CuraEngineBackend/CuraEngineBackend.py", line 158, in slice
    if kwargs.get("profile", self._profile).hasErrorValue():
  File "/usr/lib/python3/dist-packages/UM/Settings/Profile.py", line 131, in hasErrorValue
    Logger.log("w", "The setting %s has an invalid value of %s",key)
  File "/usr/lib/python3/dist-packages/UM/Logger.py", line 27, in log
    filled_message = message % args # Replace all the %s with the variables. Python formating is magic.
TypeError: not enough arguments for format string
```

nallath, 2015-09-19:
Someone on the forums already reported this. I've already fixed it in 15.10 branch.

closed 2015-09-19 by nallath

thopiekar, 2015-09-19:
Sorry, didn't know that. Just found that while playing with the UI.

nallath, 2015-09-19:
Still good that you reported it :) Rather have it one time too much

thopiekar, 2015-09-19:
No problem. I would even close this issue on my own until it's fixed (like
a todo list). Using nightly builds using my PPA here, so would notice after
24h when a new build is done and it is fixed.

Regards and thank you for caring about this problem.