Ultimaker/Cura #408: [master] cura.log created in $(pwd)

thopiekar, 2015-09-19:
While working on #407 I noticed Cura always creates it's log at $(pwd), but according to https://github.com/Ultimaker/Cura#logging-issues it should be $HOME/.cura/cura.log instead.

thopiekar, 2015-09-19:
Just adding it here from the discussion at #407:
```I don't know where you store all the settings on Linux, but $HOME/.cura even doesn't exist here :confused: Maybe it would be better to keep settings + logs at one place?```

awhiemstra, 2015-09-21:
The config file is stored at .config/cura, the other files are stored in ~/.local/share/cura, as per xdg guidelines. In the case of the log file, we should probably store it in ~/.local/share/cura with the other data files.

thopiekar, 2015-09-21:
OK, thought it was something like ~/.config/ or ~./cache/ were data should
be stored. I'm not familiar with xdg specification, so I agree with that
:+1:

TMHogenhout, 2015-10-19:
fixed by 

https://github.com/Ultimaker/Uranium/commit/5d76a84e8724d5a184fb2f73e4484531043d2065

closed 2015-10-19 by TMHogenhout