Ultimaker/Cura #409: [master] no slicing when opening file with cura

thopiekar, 2015-09-19:
As reported at #407 I noticed that slicing and generating layers are not working when opening the stl file with Cura. Using Cura's own open button slices the following stl file and of course creates layers for the layer view as expected.

nallath, 2015-09-19:
I believe this is also a bug with the 15.10 branch

thopiekar, 2015-09-19:
This can be correct, as I also noticed that in the past, when I ran cura_all.py from another working directory. I always thought this is the correct behaviour, but after my work on stl files I noticed it there, too. Looks like the working directory is set to "os.path.split(__file__)[0]" (talking in python here), when cura_app.py is executed.

closed 2015-09-22 by nallath