Ultimaker/Cura #412: 15.09.83: Printer and Profile not saved

CCS86, 2015-09-21:
I launched .83 for the first time (with a clean  install). I created a new printer, named it, customized a profile by enabling setting visibility and changing settings. Eventually, Cura crashed on a reslice, and when I relaunched. it was like a fresh install again.

I would expect the printer and profile changes to be saved on the fly, and available after a crash.

CCS86, 2015-09-22:
Hmmm, Seems like more of a bug to me.

I tried again today, with no model loaded, enabling all the settings I want visible. Then I customized most of the settings and renamed & exported my profile. I closed Cura, relaunched, then had to choose advanced again, select my machine, select my profile... and none of the settings visibility stuck. Right back to default.

CCS86, 2015-09-22:
On top of that, it didn't actually export my profile. No file was created.

When I relaunched, my profile was there (in name), but selecting it showed that none of my actual settings were saved either.

CCS86, 2015-09-22:
After choosing all my custom settings again, I tried to go into manage profiles and got a windows "cura has stopped working".

I think this is all related to issues in managing these configuration files. Hopefully this log data helps shed light on it:

2015-09-21 19:18:32,178 - INFO - Loaded plugin FileLogger
2015-09-21 19:18:32,194 - INFO - Loaded plugin GCodeWriter
2015-09-21 19:18:32,209 - INFO - Loaded plugin LayerView
2015-09-21 19:18:32,225 - INFO - Loaded plugin LocalFileOutputDevice
2015-09-21 19:18:32,241 - INFO - Loaded plugin PostProcessingPlugin
2015-09-21 19:18:32,256 - INFO - Loaded plugin RemovableDriveOutputDevice
2015-09-21 19:18:32,303 - INFO - Loaded plugin SliceInfoPlugin
2015-09-21 19:18:32,319 - INFO - Loaded plugin CameraTool
2015-09-21 19:18:32,343 - INFO - Loaded plugin MirrorTool
2015-09-21 19:18:32,352 - INFO - Loaded plugin PerObjectSettingsTool
2015-09-21 19:18:32,399 - INFO - Loaded plugin RotateTool
2015-09-21 19:18:32,414 - INFO - Loaded plugin ScaleTool
2015-09-21 19:18:32,445 - INFO - Loaded plugin SelectionTool
2015-09-21 19:18:32,477 - INFO - Loaded plugin TranslateTool
2015-09-21 19:18:32,492 - INFO - Checking for new version of cura
2015-09-21 19:18:32,492 - INFO - Loaded plugin UpdateChecker
2015-09-21 19:18:32,539 - INFO - Loaded plugin USBPrinting
2015-09-21 19:18:32,570 - INFO - Loaded plugin MeshView
2015-09-21 19:18:32,602 - INFO - Loaded plugin WireframeView
2015-09-21 19:18:32,649 - INFO - Started engine process: C:\Program Files (x86)\Cura_15.09.83\CuraEngine.exe
2015-09-21 19:18:32,742 - ERROR - An exception occurred loading Machine Instance: C:\Users\Admin/AppData/Local/cura\machine_instances\Ultimaker86.cfg: File C:\Users\Admin/AppData/Local/cura\machine_instances\Ultimaker86.cfg is an invalid settings file
2015-09-21 19:18:32,742 - ERROR - An exception occurred loading Profile C:\Users\Admin/AppData/Local/cura\profiles\86.cfg: File C:\Users\Admin/AppData/Local/cura\profiles\86.cfg is an invalid settings file
2015-09-21 19:18:32,742 - DEBUG - Setting value of setting layer_height_0 to 0.2
2015-09-21 19:18:32,742 - WARNING - Tried to set value of non-user setting layer_height_0
2015-09-21 19:18:32,742 - DEBUG - Setting value of setting cool_fan_full_layer to 3
2015-09-21 19:18:32,742 - WARNING - Tried to set value of non-user setting cool_fan_full_layer
2015-09-21 19:18:32,742 - DEBUG - Setting value of setting speed_travel to 190
2015-09-21 19:18:32,742 - WARNING - Tried to set value of non-user setting speed_travel
2015-09-21 19:18:32,742 - DEBUG - Setting value of setting travel_compensate_overlapping_walls_enabled to True
2015-09-21 19:18:32,742 - WARNING - Tried to set value of non-user setting travel_compensate_overlapping_walls_enabled
2015-09-21 19:18:32,742 - DEBUG - Setting value of setting skin_outline_count to 1
2015-09-21 19:18:32,742 - WARNING - Tried to set value of non-user setting skin_outline_count
2015-09-21 19:18:32,742 - DEBUG - Setting value of setting z_seam_type to back
2015-09-21 19:18:32,742 - WARNING - Tried to set value of non-user setting z_seam_type
2015-09-21 19:18:32,742 - DEBUG - Setting value of setting coasting_enable to True
2015-09-21 19:18:32,742 - WARNING - Tried to set value of non-user setting coasting_enable
2015-09-21 19:18:32,742 - DEBUG - Setting value of setting retraction_amount to 1
2015-09-21 19:18:32,742 - WARNING - Tried to set value of non-user setting retraction_amount
2015-09-21 19:18:32,742 - DEBUG - Setting value of setting adhesion_type to brim
2015-09-21 19:18:32,742 - WARNING - Tried to set value of non-user setting adhesion_type
2015-09-21 19:18:32,742 - DEBUG - Setting value of setting speed_wall_x to 40
2015-09-21 19:18:32,742 - WARNING - Tried to set value of non-user setting speed_wall_x
2015-09-21 19:18:32,742 - DEBUG - Setting value of setting layer_height to 0.2
2015-09-21 19:18:32,742 - WARNING - Tried to set value of non-user setting layer_height
2015-09-21 19:18:32,742 - DEBUG - Setting value of setting wall_line_count to 5
2015-09-21 19:18:32,742 - WARNING - Tried to set value of non-user setting wall_line_count
2015-09-21 19:18:32,742 - DEBUG - Setting value of setting retraction_prime_speed to 12
2015-09-21 19:18:32,742 - WARNING - Tried to set value of non-user setting retraction_prime_speed
2015-09-21 19:18:32,742 - DEBUG - Setting value of setting retraction_count_max to 3
2015-09-21 19:18:32,742 - WARNING - Tried to set value of non-user setting retraction_count_max
2015-09-21 19:18:32,742 - DEBUG - Setting value of setting cool_lift_head to True
2015-09-21 19:18:32,742 - WARNING - Tried to set value of non-user setting cool_lift_head
2015-09-21 19:18:32,742 - DEBUG - Setting value of setting remove_overlapping_walls_enabled to True
2015-09-21 19:18:32,742 - WARNING - Tried to set value of non-user setting remove_overlapping_walls_enabled
2015-09-21 19:18:32,742 - DEBUG - Setting value of setting retraction_hop to .1
2015-09-21 19:18:32,742 - WARNING - Tried to set value of non-user setting retraction_hop
2015-09-21 19:18:32,742 - DEBUG - Setting value of setting support_enable to False
2015-09-21 19:18:32,742 - WARNING - Tried to set value of non-user setting support_enable
2015-09-21 19:18:32,742 - DEBUG - Setting value of setting speed_wall_0 to 40
2015-09-21 19:18:32,742 - WARNING - Tried to set value of non-user setting speed_wall_0
2015-09-21 19:18:32,742 - DEBUG - Setting value of setting layer_height to 0.06
[... cut ...]
2015-09-21 19:20:08,101 - ERROR - Failed to write profile to C:\Users\Admin/AppData/Local/cura\profiles\86.cfg: [WinError 5] Access is denied: 'C:\\Users\\Admin\\AppData\\Local\\cura\\profiles\\tmp947o4usu' -> 'C:\\Users\\Admin/AppData/Local/cura\\profiles\\86.cfg'
2015-09-21 19:20:08,104 - ERROR - Failed to write preferences to C:\Users\Admin/AppData/Local/cura\cura.cfg: [WinError 5] Access is denied: 'C:\\Users\\Admin\\AppData\\Local\\cura\\tmp4a_dza8g' -> 'C:\\Users\\Admin/AppData/Local/cura\\cura.cfg'

CCS86, 2015-09-22:
Both the machine and profile files in the Appdata folder exist, but are completely empty

CCS86, 2015-09-22:
More pertinent log entries:

2015-09-21 20:29:14,593 - ERROR - Failed to write Instance to C:\Users\Admin/AppData/Local/cura\machine_instances\Ultimaker86.cfg: [WinError 5] Access is denied: 'C:\\Users\\Admin\\AppData\\Local\\cura\\machine_instances\\tmp58qls6_h' -> 'C:\\Users\\Admin/AppData/Local/cura\\machine_instances\\Ultimaker86.cfg'
2015-09-21 20:29:14,611 - ERROR - Failed to write profile to C:\Users\Admin/AppData/Local/cura\profiles\86.cfg: [WinError 5] Access is denied: 'C:\\Users\\Admin\\AppData\\Local\\cura\\profiles\\tmpcbvy6h6p' -> 'C:\\Users\\Admin/AppData/Local/cura\\profiles\\86.cfg'
2015-09-21 20:29:14,613 - ERROR - Failed to write preferences to C:\Users\Admin/AppData/Local/cura\cura.cfg: [WinError 5] Access is denied: 'C:\\Users\\Admin\\AppData\\Local\\cura\\tmpe2e31m6_' -> 'C:\\Users\\Admin/AppData/Local/cura\\cura.cfg'

I'm curious why the slashes bounce between back and forward slash.

Cura has been successfully writing this data to temp files in the same directory. Copy and pasting the data into the Cura.cfg file avoided the changelog and displayed my visible settings.

However, doing the same to my machine and profile, did not work:

2015-09-21 20:30:34,088 - ERROR - An exception occurred loading Machine Instance: C:\Users\Admin/AppData/Local/cura\machine_instances\Ultimaker86.cfg: File C:\Users\Admin/AppData/Local/cura\machine_instances\Ultimaker86.cfg is an invalid settings file
2015-09-21 20:30:34,088 - ERROR - An exception occurred loading Profile C:\Users\Admin/AppData/Local/cura\profiles\86.cfg: File C:\Users\Admin/AppData/Local/cura\profiles\86.cfg is an invalid settings file

nallath, 2015-09-22:
I'm not quite sure which of those it is. Lets keep it at both. 

We changed the way setting files are handled. We had a lot of writing errors when multiple instances of Cura tried to write to the same file. We changed this by using temp files that are then copied. It seems like that doesn't quite work for windows.

awhiemstra, 2015-09-22:
"[WinError 5] Access is denied" is the key here, it seems Windows does not allow you to write to the config files. Any chance you can post the properties, specifically the owner and permissions, of the cura.cfg files?

fieldOfView, 2015-09-22:
I just noticed the same. A temporary file is written, and a 0 bytes cura.cfg file is created. 
I would that that Cura still has a writing lock on cura.cfg, writes to a different file, and then tries to replace the cura.cfg file with the temporary file.

fieldOfView, 2015-09-22:
Yup, the Preferences class uses the SaveFile class to store cura.cfg, and that class places a lock on the intended file, then saves to a temp file, and tries to overwrite the locked file with the temp file. Does that work on other platforms? It doesn't on win32.

CCS86, 2015-09-23:
Any thoughts on why Cura says the files are invalid, when the temp data is pasted into the CFG files manually?

awhiemstra, 2015-09-25:
Not sure. The config loading code checks for a few keys that should exist. If an error occurs, the config file gets ignored but application startup should be fine.

CCS86, 2015-09-25:
It does launch, but after deeming the CFG files invalid, they are wiped blank, and I have to start over again configuring everything.

Here is the data I pasted into the profile:

[general]
version = 1
name = 86

[settings]
layer_height_0 = 0.2
cool_fan_full_layer = 3
speed_travel = 190
travel_compensate_overlapping_walls_enabled = True
skin_outline_count = 1
z_seam_type = back
coasting_enable = False
retraction_amount = 4
adhesion_type = brim
speed_wall_x = 40
layer_height = 0.2
wall_line_count = 5
retraction_prime_speed = 12
retraction_count_max = 3
cool_lift_head = True
remove_overlapping_walls_enabled = True
retraction_hop = .1
support_enable = False
speed_wall_0 = 40

And into the Cura.cfg:

[local_file]

[machines]
setting_visibility = layer_height,layer_height_0,wall_line_count,top_layers,bottom_layers,remove_overlapping_walls_enabled,travel_compensate_overlapping_walls_enabled,fill_perimeter_gaps,skin_outline_count,z_seam_type,infill_sparse_density,infill_pattern,material_print_temperature,material_bed_temperature,material_diameter,retraction_enable,retraction_amount,retraction_retract_speed,retraction_prime_speed,retraction_min_travel,retraction_count_max,retraction_extrusion_window,retraction_hop,speed_print,speed_wall_0,speed_wall_x,speed_travel,speed_layer_0,coasting_enable,coasting_volume,coasting_speed,cool_fan_enabled,cool_fan_full_layer,cool_min_layer_time,cool_min_speed,cool_lift_head,support_enable,support_type,support_angle,support_xy_distance,support_top_distance,support_bottom_distance,support_conical_enabled,support_conical_angle,support_conical_min_width,support_bottom_stair_step_height,support_join_distance,support_offset,support_area_smoothing,support_roof_enable,support_roof_height,support_roof_density,support_roof_pattern,support_infill_rate,adhesion_type,skirt_line_count,skirt_gap,skirt_minimal_length,brim_line_count,draft_shield_enabled,draft_shield_dist,draft_shield_height_limitation,draft_shield_height,print_sequence,magic_mesh_surface_mode,magic_spiralize
active_profile = 86
active_instance = Ultimaker86

[backend]

[cura]
active_mode = 1
recent_files = C:/Users/Admin/Desktop/3D/Models/My Designs/Chin up Grips/Ball - small.stl;

[info]
asked_send_slice_info = True

[physics]

[view]

[general]
window_left = 293
window_state = 2
window_top = 197
latest_version_changelog_shown = 15.09.83
version = 2

[mesh]

CCS86, 2015-09-25:
Hmm, the new text attachment system isn't working. Here is the cura.cfg file contents:

[local_file]

[machines]
setting_visibility = layer_height,layer_height_0,wall_line_count,top_layers,bottom_layers,remove_overlapping_walls_enabled,travel_compensate_overlapping_walls_enabled,fill_perimeter_gaps,skin_outline_count,z_seam_type,infill_sparse_density,infill_pattern,material_print_temperature,material_bed_temperature,material_diameter,retraction_enable,retraction_amount,retraction_retract_speed,retraction_prime_speed,retraction_min_travel,retraction_count_max,retraction_extrusion_window,retraction_hop,speed_print,speed_wall_0,speed_wall_x,speed_travel,speed_layer_0,coasting_enable,coasting_volume,coasting_speed,cool_fan_enabled,cool_fan_full_layer,cool_min_layer_time,cool_min_speed,cool_lift_head,support_enable,support_type,support_angle,support_xy_distance,support_top_distance,support_bottom_distance,support_conical_enabled,support_conical_angle,support_conical_min_width,support_bottom_stair_step_height,support_join_distance,support_offset,support_area_smoothing,support_roof_enable,support_roof_height,support_roof_density,support_roof_pattern,support_infill_rate,adhesion_type,skirt_line_count,skirt_gap,skirt_minimal_length,brim_line_count,draft_shield_enabled,draft_shield_dist,draft_shield_height_limitation,draft_shield_height,print_sequence,magic_mesh_surface_mode,magic_spiralize
active_profile = 86
active_instance = Ultimaker86

[backend]

[cura]
active_mode = 1
recent_files = C:/Users/Admin/Desktop/3D/Models/My Designs/Chin up Grips/Ball - small.stl;

[info]
asked_send_slice_info = True

[physics]

[view]

[general]
window_left = 293
window_state = 2
window_top = 197
latest_version_changelog_shown = 15.09.83
version = 2

[mesh]

CCS86, 2015-09-25:
And my profile:

[general]
version = 1
name = 86

[settings]
layer_height_0 = 0.2
cool_fan_full_layer = 3
speed_travel = 190
travel_compensate_overlapping_walls_enabled = True
skin_outline_count = 1
z_seam_type = back
coasting_enable = False
retraction_amount = 4
adhesion_type = brim
speed_wall_x = 40
layer_height = 0.2
wall_line_count = 5
retraction_prime_speed = 12
retraction_count_max = 3
cool_lift_head = True
remove_overlapping_walls_enabled = True
retraction_hop = .1
support_enable = False
speed_wall_0 = 40

CCS86, 2015-09-26:
This seems to be working nicely in .84!

robomaniac, 2015-09-27:
what, you got .84?

this one?
Cura-15.09.84-win64.exe	25-Sep-2015 18:36	48M	 

ok, I will download it since I have the same bug has you and I am in .83!

robomaniac, 2015-09-27:
ok it works!

awhiemstra, 2015-09-28:
Yeah sorry, there is no official announcement of .84 builds yet since I did not manage to build .84 for Win32.

CCS86, 2015-09-30:
Having issues again with 15.09.85

It seems that Cura is not using the AppData folder anymore to store CFG files. Once I copied my profile to the Cura program files \ resources directory, it read just fine. However, I can't find where it is now keeping the cura.cfg file.

Settings visibility keeps getting lost.

closed 2016-12-26 by thopiekar