Ultimaker/Cura #413: 15.09.83: Crash on reslice.

CCS86, 2015-09-21:
Not sure if there is a way to get more data on this crash. Looks like possibly an engine issue, since the last entry is sending data to slicing engine. Not sure if the other warnings are important. This crash showed up as a windows "Cura experienced and error and must close".

[image]

CCS86, 2015-09-21:
Might have been triggered by setting shell thickness to .4mm while wall line count was set to 5.

Ghostkeeper, 2015-09-22:
Definitely looks like an engine issue since the engine crashes, but that shouldn't mean that the front-end crashes too...

nallath, 2015-09-22:
Does the GUI crash? It does seem to go wrong on the engine side.

CCS86, 2015-09-23:
Yup, the GUI crashed with a windows prompt.

BagelOrb, 2015-10-01:
The end of the log says:

============================================

2015-09-21 09:03:22,628 - INFO - Backend crashed or closed. Restarting...
2015-09-21 09:03:22,935 - INFO - Started engine process: C:\Program Files (x86)\Cura_15.09.83\CuraEngine.exe
2015-09-21 09:03:22,977 - DEBUG - Backend connected on port 49674
2015-09-21 09:03:23,479 - DEBUG - Preparing to send slice data to engine.
2015-09-21 09:03:23,494 - DEBUG - Sending data to engine for slicing.

=====================================================

I don't think this bug is solvable with the limited information we have.

closed 2015-10-01 by BagelOrb

Ghostkeeper, 2015-10-01:
Since this all happened 10 days ago, it's too late to direct CCS86 to the engine log now, but the engine log might shed some light on this. If it ever happens again, we'll have to find that log.