Ultimaker/Cura #414: py2exe problems on windows 7

nbodard, 2015-09-21:
Hi all,

When I want to get the exe file with the command "python cura_app.py py2exe", I get two kinds of error:
    - with the module "CuraEngineBackend" and "google" module name,   
    - with the module "USBPrinting" and "serial" module name. 

Could you help me to detect the reasons why these errors occurs dispite i think nothing is missing in my installation?
Is there something to modify with the config.cfg file and should I have to install this file in windows 7 in a special directory? 

Best regards,
Nicolas

...
2015-09-21 12:59:29,017 - ERROR - Import error loading module USBPrinting: No module named 'serial'
2015-09-21 12:59:29,019 - ERROR - Could not find plugin USBPrinting
2015-09-21 12:59:29,032 - ERROR - Import error loading module USBPrinting: No module named 'serial'
...
2015-09-21 12:59:29,088 - ERROR - Import error loading module CuraEngineBackend: No module named 'google'
2015-09-21 12:59:29,090 - ERROR - Could not find plugin CuraEngineBackend
2015-09-21 12:59:29,095 - ERROR - Import error loading module CuraEngineBackend: No module named 'google'
...
Version: 15.09.82
Platform: Windows-7-6.1.7601-SP1
Qt: 5.5.0
PyQt: 5.5

Exception:
Traceback (most recent call last):
  File "cura_app.py", line 16, in <module>
    app = cura.CuraApplication.CuraApplication.getInstance()
  File "C:\Python34\lib\site-packages\UM\Application.py", line 273, in getInstance
    Application._instance = cls()
  File "C:\Python34\lib\site-packages\cura\CuraApplication.py", line 71, in __init__
    super().__init__(name = "cura", version = "15.09.82")
  File "C:\Python34\lib\site-packages\UM\Qt\QtApplication.py", line 79, in __init__
    self._loadPlugins()
  File "C:\Python34\lib\site-packages\cura\CuraApplication.py", line 129, in _loadPlugins
    raise RuntimeError("Could not load the backend plugin!")
RuntimeError: Could not load the backend plugin!

awhiemstra, 2015-09-21:
The "google" name contains protobuf which is a hard dependency. See Ultimaker/Protobuf for a forked version that is easy to build with CMake. The module "serial" is Python's PySerial. Depending on how you are building things, they either should be built/downloaded and installed manually or the build scripts should take care of it. Note that the Windows build of curs-build is not yet completely functional.

nbodard, 2015-09-21:
Hi awhiemstra,
I solved the second point with pyserial installation using pip. I tried the same thing with protobuf but it gave mistakes already with pip install. 

Do I have to use my own protobuf version i made with  "Ultimaker/Protobuf" or do I have to search a newer version of protobuf  with pip?

Thank you very much for your help.
Nicolas

awhiemstra, 2015-09-22:
You can use either the master version of protobuf or our forked repo. Both build Protobuf 3 which is what we require. According to PyPI, there is no package in PyPI (that is used by pip) that provides protobuf 3 for Python 3. So you will need to build your own I'm afraid. Do note that you will need protobuf anyway for the engine to build.

nbodard, 2015-09-22:
Thank you awhiemstra,

I found a solution adding a correct version of protobuf with a correct modification between python 2 and python 3 syntaxe.

closed 2015-09-22 by nallath