Ultimaker/Cura #416: 15.09.83: Hang on 2nd application load

CCS86, 2015-09-22:
On a fresh install of .83, I launched, added my machine, and decided to exit and relaunch to make sure it was being saved. 

When I relaunched, the main window and the changelog window appeared (again), but both were all white and unresponsive. They never recover.

After forcing it to close, I can successfully relaunch.

Running on Windows 10 10240

awhiemstra, 2015-09-25:
Did you manage to reproduce this issue? Because I do not see it happening here.

robomaniac, 2015-09-27:
I tried what you said,

- I added new printer "ultimaker 2 go"
- then close Cura.
- I open Cura (The change log for 15.10 appears) yes it's annoying :)
- Then in the right pane, the print setup is "simple" and the machine dropdown menu is white (Not selected)

There is nothing seleced in the Machine. I click the dropdown menu and select "ultimaker 2 go"
Please do a print screen and drag and drop inside the comment section it will automatically upload.

CCS86, 2015-09-27:
I reproduced it today in .84

It wasn't on the second launch, but after many. This screenshot shows what the Cura window looks like, and also where it stopped logging:

[image]

closed 2016-12-26 by thopiekar