Ultimaker/Cura #418: Slicing starts at each change of position while holding the button

iakapulko, 2015-09-22:
During the rotation of the object along the axes, slicing runs each time you change the degree of tilt. All this time the mouse button is pressed. The log is no issue, but i can make a small video, if necessary.

iakapulko, 2015-09-22:
Update: During the rotation at the bottom there is the inscription "Slicing", then within a second start slicing (percentage of completion). Even if changes were not made (+15 degrees, and return to the original state in one clamping mouse) slicing still starts.

nallath, 2015-09-22:
Videos are always good :) helps us a lot.

iakapulko, 2015-09-22:
https://www.dropbox.com/s/zkybc4wbq8ua5eo/cura%20slicing.exe?dl=0 Play with a fullscreen mode.

awhiemstra, 2015-09-22:
Hmm, we postpone platform physics processing while performing an operation, but not slicing it seems. It should be fairly simple to fix.

nallath, 2015-09-22:
That was a really, really silly bug. All the implementation was there, it just wasn't called. Ever.

closed 2015-09-22 by nallath