Ultimaker/Cura #419: py2exe windows 7 stale

nbodard, 2015-09-22:
Hi All,

After some modifications and using "python cura_app.py py2exe", I got  following the message at the end with a changelog cura windows with 15.10.0 on top:
....
2015-09-22 09:24:40,254 - DEBUG - Setting active view to MeshView
QQmlApplicationEngine failed to load component
file:///C:/Cura/share/cura/resources/qml/Cura.qml:18 Cannot assign to non-existent property "viewportRect"
2015-09-22 09:24:41,471 - WARNING - Unable to read file C:\Cura\bin\py2exe
2015-09-22 09:24:41,500 - DEBUG - Preparing to send slice data to engine.

Why this kind of message occurs and why the system remains staled showing a  changelog windows with version 15.10.0 ?

nallath, 2015-09-22:
This is an issue due to old versions of Uranium still being installed. The old version of Uranium did nog have the property viewportRect, but the latest version does (hence the conflict)

awhiemstra, 2015-09-22:
Note that calling "python cura_app.py py2exe" will try to open the "py2exe" file, not run py2exe to convert cura_app.py to an executable. But if you're not building a package, there is no real need to run py2exe.

nbodard, 2015-09-22:
Hi Nallath, hi Awhiemstra,

Thank you for your remarks. I had a too old version of Uranium. Indeed, "py2exe" option doesn't work because it tried to read the file py2exe. I ran simply   "python cura_app.py" and the application is finally launched but I don't get an exe file. I can go further now. If you know how i can get an exe file, I am still open. 
 
Again, thank you very much,
Nicolas

awhiemstra, 2015-09-22:
I'd recommend waiting for the cura-build Windows work to be finished for building executable files. It should be done this week and will make it a lot easier to get a working Windows executable.

(The current way I build Windows builds is to use the build.sh script, but it needs heavy, dev environment specific, modifications to actually work properly, which is why I started work on a replacement.)

awhiemstra, 2015-09-25:
cura-build now works for building 32-bit executables on Windows. While the build instructions need a bit of updating the basics work so please see that repository if you want to build executables.

closed 2015-09-25 by awhiemstra