Ultimaker/Cura #421: uncaught exception after install version 15.06.03

rdreilly, 2015-09-22:
Version: 15.06.03
Platform: Windows-7-6.1.7601-SP1
Qt: 5.4.1
PyQt: 5.4.1

Exception:
Traceback (most recent call last):
  File "cura_app.py", line 9, in <module>
  File "c:\dev\uranium\UM\Application.py", line 329, in getInstance
  File "c:\dev\cura\cura\CuraApplication.py", line 56, in __init__
  File "c:\dev\uranium\UM\Qt\QtApplication.py", line 48, in __init__
  File "c:\dev\uranium\UM\Application.py", line 67, in __init__
  File "c:\dev\uranium\UM\Preferences.py", line 56, in readFromFile
  File "c:\dev\uranium\UM\Preferences.py", line 124, in _loadFile
  File "c:\dev\python34\lib\configparser.py", line 937, in __getitem__
KeyError: 'general'

Ghostkeeper, 2015-09-22:
Duplicate of #72, #104, #199, #246, #275, #344 and #349. Something went wrong when saving the configuration file. The solution is to delete the configuration file, located in %APPDATA/local/cura/cura.cfg. This should trigger Cura to re-make that file, but you'll lose your preferences.

This has been fixed for version 15.10.

closed 2015-09-22 by Ghostkeeper

thopiekar, 2015-09-23:
What about catching these errors and ask the user whether to delete it?
Definitely it would save the users time to search for the configuration file.
Additionally it would be helpful for your to send this configuration file via user statics feature to understand the problem. Just deleting might not solve problems with it in the future.

nallath, 2015-09-23:
We have an actual fix for this for 15.10, which should prevent the issue. We use the preferences in too many places to simply add exception catching to all of them. That and i'd rather not spend more time in fixing an old version, especially if we have a 'real' fix for the next one.

thopiekar, 2015-09-23:
OK then. :)

rdreilly, 2015-09-23:
Thomas-Karl, Jaime

Thank you for your hard work and quick response on this issue.

 

bob

 

Bob Reilly

Senior Software Engineer

3D Imaging &Simulations Corp. (3DISC)

22560 Glenn Drive, Suite 116

Sterling VA 20164

United States

Tel: (+1) 800 570 0363

Fax: (+1) 703 430 8320

Cell: (+1) 571 217 9834

E-mail: bre@3DISCimaging.com <mailto:bre@3DISCimaging.com>  

Website:  <http://www.3discimaging.com/> www.3DISCimaging.com

 

 <http://www.linkedin.com/company/3disc-imaging>     <https://www.facebook.com/3DISCimaging>     <https://twitter.com/3DISCimaging>     <http://www.youtube.com/3DISCimaging>