Ultimaker/Cura #422: Propose on modification of the Cura's GUI

iakapulko, 2015-09-23:
As suggested @TMHogenhout, spread variant of the interface, which seems to me close to ideal. Changes are described in the callouts. If the dwg format is not very convenient, can do the export to pdf.

iakapulko, 2015-09-23:
In png format.
[image]

nallath, 2015-09-23:
I agree with the moving of the slicing message. We used to have the slicing in the place where you put it, but for some reason we moved it, but I find the popping op of the message all the time rather annoying.

iakapulko, 2015-09-23:
Annoyed is quite an understatement. Why multiply entities? And so, the user first, will have a unified place for the progress bar, and secondly, he will not be pressing that button until the moment when a strip is filled on 100%. And Yes, this progress bar in a corner, and there IS a button.

nallath, 2015-09-23:
@mennoberg What do you think?

mennoberg, 2015-09-23:
@iakapulko thanks for valuable feedback,I can see what your problem is. This also imo an issue because your focus, control and view on the 3d platform is not in balance with the appearance of the tools. Our focus is to improve this first and then consider the position of the tools again. Proposal are ready for this, @awhiemstra told me that this is on its list. 
@iakapulko Any feedback or input on use and control of the 3d buildplate?

iakapulko, 2015-09-23:
@mennoberg, always please, if it will benefit the program in the end. What I miss when working with buildplate? Well, there is one very useful feature in autocad that is often used by me: "видовой куб" in russian version of autocad. I dont know, how to correct translate this in english, but maybe attached picture can help you, what i mean.
[image]

iakapulko, 2015-09-23:
What is even more useful in it, is the presence of the "home" button that returns the display angle of the model to default state.

nallath, 2015-09-23:
Old cura has that if you press the home button on your keyboard.

iakapulko, 2015-09-23:
15.10 have the same keyboard layont?

nallath, 2015-09-23:
Nope, home button doesn't work there. But if people would like it, it should be relatively easy to implement I guess.

mennoberg, 2015-09-23:
I made an example of that in a newer version Im working on. Will show u asap!

nallath, 2015-09-23:
Nice :)

iakapulko, 2015-09-24:
Thats great!

mennoberg, 2015-09-24:
The view cube your proposed is there in Autodesk for the user to relate to a certain orientation of your view perspective. In Cura you can relate to the 3D buildplate and how that matches visually with your Ultimaker 2 buildplate (skeuomorphism) and in that way helps your orientate your perspective view. So somehow this whole cube could be double and therefor unnecessary. Though I can see that people would like to use functions as top view or side view. Where the view orientation instantly moves to this perspective. As well as it gives some trust in how you navigate in this 3D space. Especially to people that are not used to this form of navigation, those might be handy and welcome functions.

iakapulko, 2015-09-24:
The proposal to introduce keyboard shortcuts shifting the view angle at fixed values of the axes (90/180/360/home) sounds great.

Ghostkeeper, 2015-09-24:
The other things proposed here are probably fine, but we have to be careful with copying the viewcube from Autodesk because I can imagine them having some sort of patent on that.

iakapulko, 2015-09-24:
@Ghostkeeper right. 
http://www.4-traders.com/AUTODESK-INC-4937/news/Autodesk--Patent-Issued-for-Configurable-Viewcube-Controller-20485105/ 
Viewcube is patented. Sad, but true.

Ghostkeeper, 2015-09-24:
Whoa they actually did, too. Thanks for finding that!

fieldOfView, 2015-09-25:
> We used to have the slicing in the place where you put it, but for some reason we moved it,

That would be "my fault"; I pushed for having only one place for progress bars. There used to be progress bars in the message stack AND a slicing progressbar in the pane. I argued for a single area to show progress.

Showing progress only in the pane means that you can only show one process in progress at a time. I can't come up with a good example in Cura where you would have multiple processes going on at the same time though.

fieldOfView, 2015-09-25:
@iakapulko What I don't like about your sketch is that it spreads controls out to the four corners of the screen. Spreading them out like that adds a considerable effort for the user to re-focus and "re-aim".

awhiemstra, 2015-09-25:
"Showing progress only in the pane means that you can only show one process in progress at a time. I can't come up with a good example in Cura where you would have multiple processes going on at the same time though. "

I can come up with several:

* Loading multiple files
* Saving GCode to a file while printing via USB (Currently uses a dialog but I want to use a message for it)
* Loading a file while slicing is happening in the background.
* Updating the layer view while slicing.
* Generally, any longer running task while slicing in the background.

In general, while moving the slicing progress back to the save button is probably a good idea, I would not move all progress there. Many of those are actually user initiated tasks however, so I do not think it is necessarily a bad idea to have a bit of difference there.

nallath, 2015-09-25:
So @fieldOfView, what is your opinion on moving the slicing back to the save button? I'm all in favour for that (but leaving loading & building layers in a message). I think that way a lot of people will have much less issues with the auto slicing. It's rather annoying to have the message pop up and down again.

fieldOfView, 2015-09-25:
@nallath: Allow me to twist and use your own reasoning against you:
If (seeing) the frequent starting and stopping of the slicing process annoys people, wouldn't it be better to fix that, than to hide the progress away?

I am still in favor of having progress shown in one location, instead of several disjointed locations. The user should not have to know how the program is structured to know where to look for progress (Cura vs Cura Engine progress). Showing all progress in the lower right is not a good idea IMHO: I don't think showing eg the progress of loading a file in the lower right of the screen "connects" with the load object button being in the top left.

So, my opinion is that the autoslicing should be made less annoying.

Looking at other applications for inspiration is always a good idea. In Photoshop for example, you only get a progressbar once a process has taken one or two seconds. This could help in having fewer "popups", when the next auto-slicing iteration is started within the "pause-before-showing-progress-period" (though, yes, this is actually hiding "the problem").

iakapulko, 2015-09-25:
@fieldOfView The analogy with photoshop is clear. But isn't it easier then to create the panel, which the user can attach to the side of the screen where it will be beneficial to him?
@awhiemstra It bothers me the idea to notify the user within a second or two that's obvious. Explain: when loading multiple models at the same time, they will not appear on screen until they are loaded into the program memory to 100%. After that, automatically start slicing. I think not much mistaken, but loading and working with multiple models in this case and so will be good to overload the system, as data of message, due to different hardware configurations, slow down your computer and annoy users.
Also not particularly clear the following: why each time you start to display a message about gathering statistics? If the user turned it off during the first launch (the fear of Big Brother who is watching you?) then why show it again and again? Why do we need a message that the file is loaded? I can't imagine a situation where the file does not load without hanging the program or operating system?
What should be a users case for these pop-up messages really necessary?

nallath, 2015-09-25:
I don't think we should hide all settings in the right corner, just the slicing one. As the slicing in progress actually blocks the save button it makes a lot of sense to put it back into the save button (as it gives a very good indication at what exact point the button starts working again).

iakapulko, 2015-09-25:
@nallath Hivemaind :)

awhiemstra, 2015-09-25:
"Also not particularly clear the following: why each time you start to display a message about gathering statistics?"

That one should be hidden if you click "dismiss". If it is not, that is a bug.

"Why do we need a message that the file is loaded? I can't imagine a situation where the file does not load without hanging the program or operating system?"

The loading message indicates something is going on in the background. The "loaded" message is simply the finalization of that process. Maybe its timeout should be tweaked a little, I do think it is useful to indicate the process ended.

fieldOfView, 2015-09-25:
One of the problems of the slice progress is that it has two different "workflow modes" that come one after the other. On the one hand it is a background-process that you can/should ignore (while making changes); showing it too much is deemed "annoying". 

On the other hand it is a blocking process; not in computer-science terms, but in workflow-terms. At some point the user is going to sit and wait for the slice to complete, before he/she can continue to save the file etc.

Is this enough of a difference to separate out the one progress bar from all the others? I'm not sure.
If slicing-progress is moved back to the output-area of the settings pane, then where does the progress of loading the layers into layer viewmode go? Or should all processes have their own progressbars? (eg a load progress-bar next to the open model button).

I do agree to an extend that because the slicing-progress is currently shown the same way as workflow-blocking operations like loading an object, some users will see it as a workflow-blocking operation even if it is not (ie: while changing settings).

fieldOfView, 2016-11-09:
This issue is ancient history and can be closed

closed 2016-11-09 by nallath