Ultimaker/Cura #427: [15.10 + master] Segfault when closing application

thopiekar, 2015-09-24:
Just found this when opening a file via konsole:

```
thopiekar@home:~/Downloads/3D_Models$ cura_app.py teapot.obj 
2015-09-24 16:17:42,653 - INFO - Loaded plugin ConsoleLogger
2015-09-24 16:17:42,660 - INFO - Loaded plugin LocalFileOutputDevice
2015-09-24 16:17:42,663 - INFO - Loaded plugin STLWriter
2015-09-24 16:17:42,666 - INFO - Loaded plugin OBJReader
2015-09-24 16:17:42,669 - INFO - Loaded plugin OBJWriter
2015-09-24 16:17:42,672 - INFO - Loaded plugin STLReader
2015-09-24 16:17:42,680 - INFO - Loaded plugin 3MFWriter
2015-09-24 16:17:42,680 - WARNING - Plugin ConsoleLogger was already loaded
2015-09-24 16:17:42,685 - INFO - Loaded plugin TranslateTool
2015-09-24 16:17:42,689 - INFO - Loaded plugin SelectionTool
2015-09-24 16:17:42,695 - INFO - Loaded plugin MirrorTool
2015-09-24 16:17:42,699 - INFO - Loaded plugin CameraTool
2015-09-24 16:17:42,705 - INFO - Loaded plugin PerObjectSettingsTool
2015-09-24 16:17:42,713 - INFO - Loaded plugin ScaleTool
2015-09-24 16:17:42,731 - INFO - Loaded plugin RotateTool
2015-09-24 16:17:42,735 - INFO - Loaded plugin MeshView
2015-09-24 16:17:42,738 - INFO - Loaded plugin WireframeView
2015-09-24 16:17:42,741 - INFO - Loaded plugin FileLogger
2015-09-24 16:17:42,799 - INFO - Loaded plugin SliceInfoPlugin
2015-09-24 16:17:42,808 - INFO - Loaded plugin USBPrinting
2015-09-24 16:17:42,815 - INFO - Loaded plugin 3MFReader
2015-09-24 16:17:42,820 - INFO - Loaded plugin RemovableDriveOutputDevice
2015-09-24 16:17:42,902 - INFO - Loaded plugin CuraEngineBackend
2015-09-24 16:17:42,906 - INFO - Loaded plugin LayerView
2015-09-24 16:17:42,910 - INFO - Loaded plugin ChangeLogPlugin
2015-09-24 16:17:42,914 - INFO - Loaded plugin GCodeWriter
2015-09-24 16:17:42,918 - INFO - Loaded plugin GCodeReader
2015-09-24 16:17:43,077 - DEBUG - Setting value of setting support_roof_enable to True
2015-09-24 16:17:43,078 - DEBUG - Setting value of setting layer_height to 0.06
2015-09-24 16:17:43,078 - DEBUG - Setting value of setting layer_height to 0.1
2015-09-24 16:17:43,079 - DEBUG - Setting value of setting layer_height to 0.06
2015-09-24 16:17:43,124 - DEBUG - Setting active view to MeshView
2015-09-24 16:17:43,260 - DEBUG - Calculating normals took 0.004913806915283203 seconds
2015-09-24 16:17:43,260 - DEBUG - Loaded a mesh with 11832 vertices
2015-09-24 16:17:43,283 - INFO - Started engine process: /usr/bin/CuraEngine
2015-09-24 16:17:43,980 - DEBUG - Loaded theme /usr/share/cura/resources/themes/cura
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/Menu.qml:146: TypeError: Cannot read property '__contentItem' of null
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/Menu.qml:146: TypeError: Cannot read property '__contentItem' of null
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/Menu.qml:146: TypeError: Cannot read property '__contentItem' of null
2015-09-24 16:17:47,131 - DEBUG - Backend connected on port 49674
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/Styles/Base/ButtonStyle.qml:154: TypeError: Cannot read property of null
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/Styles/Base/ButtonStyle.qml:154: TypeError: Cannot read property of null
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/Styles/Base/ButtonStyle.qml:154: TypeError: Cannot read property of null
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/Styles/Base/ButtonStyle.qml:154: TypeError: Cannot read property of null
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/Styles/Base/ButtonStyle.qml:154: TypeError: Cannot read property of null
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/Styles/Base/ButtonStyle.qml:154: TypeError: Cannot read property of null
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/Styles/Base/ButtonStyle.qml:154: TypeError: Cannot read property of null
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/Styles/Base/ButtonStyle.qml:154: TypeError: Cannot read property of null
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/Styles/Base/ButtonStyle.qml:154: TypeError: Cannot read property of null
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/Styles/Base/ButtonStyle.qml:154: TypeError: Cannot read property of null
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/Styles/Base/ButtonStyle.qml:154: TypeError: Cannot read property of null
2015-09-24 16:17:47,271 - DEBUG - Preparing to send slice data to engine.
2015-09-24 16:17:47,379 - DEBUG - Calculating normals took 0.005791664123535156 seconds
2015-09-24 16:17:47,389 - DEBUG - Loading mesh took 0.15824007987976074 seconds
2015-09-24 16:17:48,101 - DEBUG - Preparing to send slice data to engine.
2015-09-24 16:17:48,121 - DEBUG - Sending data to engine for slicing.
Speicherzugriffsfehler (Speicherabzug geschrieben)
```

awhiemstra, 2015-09-25:
What exact platform are you running on? Specifically, what DE? Because I know of a crash on close with KDE due to the Breeze style.

thopiekar, 2015-09-25:
You exactly suggested my DE.
Kubuntu wily with KDE Plasma on amd64 Linux using "nvidia" drivers.

awhiemstra, 2015-09-25:
If you can provide a backtrace I can check to see if it is the same, but it probably is. It seems to be fine with breeze-5.4.1 here so this might already be fixed upstream.

thopiekar, 2015-09-25:
```
GNU gdb (Ubuntu 7.10-1ubuntu2) 7.10
Copyright (C) 2015 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from python3...Reading symbols from /usr/lib/debug//usr/bin/python3.4...done.
done.
(gdb) handle SIG33 pass nostop noprint
Signal        Stop	Print	Pass to program	Description
SIG33         No	No	Yes		Real-time event 33
(gdb) set pagination 0
(gdb) run /usr/bin/cura_app.py
Starting program: /usr/bin/python3 /usr/bin/cura_app.py
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffedd51700 (LWP 4772)]
[New Thread 0x7fffed550700 (LWP 4773)]
[New Thread 0x7fffead4f700 (LWP 4774)]
[New Thread 0x7fffd9e23700 (LWP 4775)]
2015-09-25 11:31:27,651 - INFO - Loaded plugin ConsoleLogger
2015-09-25 11:31:27,656 - INFO - Loaded plugin LocalFileOutputDevice
2015-09-25 11:31:27,659 - INFO - Loaded plugin STLWriter
2015-09-25 11:31:27,663 - INFO - Loaded plugin OBJReader
2015-09-25 11:31:27,665 - INFO - Loaded plugin OBJWriter
2015-09-25 11:31:27,668 - INFO - Loaded plugin STLReader
2015-09-25 11:31:27,684 - INFO - Loaded plugin 3MFWriter
2015-09-25 11:31:27,684 - WARNING - Plugin ConsoleLogger was already loaded
2015-09-25 11:31:27,690 - INFO - Loaded plugin TranslateTool
2015-09-25 11:31:27,694 - INFO - Loaded plugin SelectionTool
2015-09-25 11:31:27,700 - INFO - Loaded plugin MirrorTool
2015-09-25 11:31:27,705 - INFO - Loaded plugin CameraTool
2015-09-25 11:31:27,711 - INFO - Loaded plugin PerObjectSettingsTool
2015-09-25 11:31:27,718 - INFO - Loaded plugin ScaleTool
2015-09-25 11:31:27,737 - INFO - Loaded plugin RotateTool
2015-09-25 11:31:27,741 - INFO - Loaded plugin MeshView
2015-09-25 11:31:27,745 - INFO - Loaded plugin WireframeView
2015-09-25 11:31:27,748 - INFO - Loaded plugin FileLogger
2015-09-25 11:31:27,836 - INFO - Loaded plugin SliceInfoPlugin
[New Thread 0x7fffc8267700 (LWP 4777)]
2015-09-25 11:31:27,858 - INFO - Loaded plugin USBPrinting
2015-09-25 11:31:27,869 - INFO - Loaded plugin 3MFReader
[New Thread 0x7fffc7a66700 (LWP 4778)]
2015-09-25 11:31:27,876 - INFO - Loaded plugin RemovableDriveOutputDevice
[New Thread 0x7fffc6ea5700 (LWP 4779)]
2015-09-25 11:31:27,960 - INFO - Loaded plugin CuraEngineBackend
2015-09-25 11:31:27,964 - INFO - Loaded plugin LayerView
2015-09-25 11:31:27,969 - INFO - Loaded plugin ChangeLogPlugin
2015-09-25 11:31:27,973 - INFO - Loaded plugin GCodeWriter
2015-09-25 11:31:27,977 - INFO - Loaded plugin GCodeReader
2015-09-25 11:31:28,128 - DEBUG - Setting value of setting layer_height to 0.06
2015-09-25 11:31:28,128 - DEBUG - Setting value of setting support_roof_enable to True
2015-09-25 11:31:28,129 - DEBUG - Setting value of setting layer_height to 0.1
2015-09-25 11:31:28,129 - DEBUG - Setting value of setting layer_height to 0.06
[New Thread 0x7fffc6524700 (LWP 4780)]
[New Thread 0x7fffc5d23700 (LWP 4781)]
[New Thread 0x7fffc5522700 (LWP 4782)]
[New Thread 0x7fffc4d21700 (LWP 4783)]
2015-09-25 11:31:28,142 - DEBUG - Setting active view to MeshView
2015-09-25 11:31:28,281 - DEBUG - Calculating normals took 0.005422115325927734 seconds
2015-09-25 11:31:28,281 - DEBUG - Loaded a mesh with 11832 vertices
[... 355 lines cut ...]
#7  0x00007ffff54acd57 in ?? () from /usr/lib/nvidia-352/libGL.so.1
#8  0x00007ffff54ad898 in ?? () from /usr/lib/nvidia-352/libGL.so.1
#9  0x00007ffff3e2134c in ?? () from /usr/lib/nvidia-352/libnvidia-glcore.so.352.30
#10 0x00007ffff3e215ab in ?? () from /usr/lib/nvidia-352/libnvidia-glcore.so.352.30
#11 0x00007ffff3e21639 in ?? () from /usr/lib/nvidia-352/libnvidia-glcore.so.352.30
#12 0x00007ffff3f02cad in ?? () from /usr/lib/nvidia-352/libnvidia-glcore.so.352.30
#13 0x00007ffff3f058ac in ?? () from /usr/lib/nvidia-352/libnvidia-glcore.so.352.30
#14 0x00007ffff3ec73b3 in ?? () from /usr/lib/nvidia-352/libnvidia-glcore.so.352.30
#15 0x00007ffff3ec767c in ?? () from /usr/lib/nvidia-352/libnvidia-glcore.so.352.30
#16 0x00007ffff3e43ac6 in ?? () from /usr/lib/nvidia-352/libnvidia-glcore.so.352.30
#17 0x00007ffff3e1d29d in ?? () from /usr/lib/nvidia-352/libnvidia-glcore.so.352.30
#18 0x00007ffff3e1ed01 in ?? () from /usr/lib/nvidia-352/libnvidia-glcore.so.352.30
#19 0x00007ffff54fa1b9 in ?? () from /usr/lib/nvidia-352/libGL.so.1
#20 0x00007ffff7de9bd5 in _dl_fini () at dl-fini.c:257
#21 0x00007ffff782ad32 in __run_exit_handlers (status=0, listp=0x7ffff7bb5698 <__exit_funcs>, run_list_atexit=run_list_atexit@entry=true) at exit.c:82
#22 0x00007ffff782ad85 in __GI_exit (status=<optimized out>) at exit.c:104
#23 0x00007ffff7811a47 in __libc_start_main (main=0x4c2c50 <main>, argc=2, argv=0x7fffffffde88, init=<optimized out>, fini=<optimized out>, rtld_fini=<optimized out>, stack_end=0x7fffffffde78) at libc-start.c:323
#24 0x00000000005b82b9 in _start ()
(gdb) quit
A debugging session is active.

	Inferior 1 [process 4757] will be killed.

Quit anyway? (y or n) y
```

awhiemstra, 2015-09-25:
Hmm, not sure if it actually is the same. But it does seem to happen somewhere deep in some shutdown code.

thopiekar, 2015-09-25:
Maybe a bug in PyQt5 or even Qt5 itself?

awhiemstra, 2015-09-28:
Might be a combination of Qt + printer. If you can install debug symbols for Qt it might be possible to find out where in Qt it crashes.

thopiekar, 2015-12-22:
Looks like it has been fixed by the KDE team. Can be closed, if you like :)

Ghostkeeper, 2015-12-23:
Gonna take your word on that :wink:

closed 2015-12-23 by Ghostkeeper