Ultimaker/Cura #429: Can't add "Bed temperature" setting

fieldOfView, 2015-09-25:
The Bed temperature setting can not be added to the settings pane when the current machine is an (upgraded) UMO.

Steps to reproduce:
* remove cura profile
* start cura
* add new Ultimaker Original
* select Heated Bedkit option
* skip through the rest

Note: bed temperature is not added, even though Heated Bedkit is selected

* open machine preferences
* add the "Bed temperature" option

Note: no new option in the settings pane appears 

With all the other options, new controls are added immediately. Not sure what is different about the Bed temperature controls.

awhiemstra, 2015-09-25:
This is due to the "Select Upgrades" page not properly setting the value of the "machine_heated_bed" setting.

CCS86, 2015-09-27:
Could someone let me know what the "machine setting" line is to enable heated bed?

I have tried creating a new print of different types, and despite editing the machine resource file to change the bed flag to "true", the created machine cfg still has a blank machine settings section.

CCS86, 2015-09-27:
Looks like there are some formatting issues in the json files.

Here is an excerpt from the UMO file:

    "overrides": {
        "material_bed_temperature": { "visible": false }
    }

And from UM2:

    "overrides": {
        "material_print_temperature": { "enabled": "True" },
        "material_bed_temperature": { "enabled": "True" },
        "material_diameter": { "enabled": "False" },
        "material_flow": { "enabled": "False" }
    }

One calls for "visible" and the other for "enabled". Also, the UMO files doesn't use quotes around the value. Copying the the overrides section from UM2 to UMO and changing the values to true, then creating a new printer allowed me bed temp setting.

nallath, 2015-09-28:
Enabled means that the value is setable at all, visible if it should be visible by default (can be overriden by user).

CCS86, 2015-09-28:
If I remember right, the UM2 file had enabled set to false.

What about the use of quotations?

TMHogenhout, 2015-10-14:
With me the "machine_heated_bed" setting is set fine. But it gets overridden as soon as you add another UM orginal without a heated bed. 

Here's what I did: 
Add an UMO with extruder driver upgrades and a heated bed. Close the program. This is a screenshot of the machine instanceses config file 
[image]

So the heated bed is added but you're missing the extruder driver upgrades right?Now add an UMO without extruder driver upgrades and without a heated bed. Screenshots of the same config file: 
[image]
After I selected reload: 
[image]

TMHogenhout, 2015-10-14:
f9f0969a6d2ef1ab348ac57e784a6cff65c1ae50

TMHogenhout, 2015-10-14:
d9041db6c9be7be0ccbf32791ba4d0b81801e393

TMHogenhout, 2015-10-14:
The issue remains that it is impossible for a user to have multiple UM Originals with slightly different upgrades. I've already checked and this was possible in the old Cura. @awhiemstra Do you consider this an issue?

awhiemstra, 2015-10-14:
The issue is not the overrides of the values. The issue here is that you use the same name for both machines, so the last one that is written writes the actual values. We should handle that better, the only issue remaining for this one is that you need to restart Cura for the changed setting to apply.

TMHogenhout, 2015-10-15:
I'll create a new issue

closed 2015-10-15 by TMHogenhout