Ultimaker/Cura #431: Reset setting does not work on checkboxes

fieldOfView, 2015-09-25:
I can't reopen #396, so I'm opening a new issue.

Steps to reproduce:
* (un)check a checkbox value (eg Enable retraction) => reset control (backward spinning icon) is added
* Press reset control

Expected result:
Reset control disappears, checkbox is reset to its original value

Actual result:
Reset control disappears, checkbox is unaffected. Manually resetting the checkbox to its original value to original setting makes the reset control reappear (though it can be removed again by pressing it).

NB: when a checkbox value is changed, the application is closed and reopened, and the reset control is pressed, it seems to work as intended. So the bug may not actually be in the reset button, but in the checkbox not setting/storing a value until the application is closed. Or something

awhiemstra, 2015-09-25:
You can actually see that it is just a display issue when changing settings like "Enable Support". The visibility of related settings changes correctly but the check mark does not update.

Additionally, this is also an issue with combo boxes.

CCS86, 2015-09-27:
I have noticed this on some of the numeric fields. 

For example, if you have displayed outer and inner loop speed, and clicked to reset both, I think they should stay associative to the parent print speed setting.

closed 2015-10-16 by awhiemstra