Ultimaker/Cura #432: SD Card Path save

LilBub, 2015-09-25:
As it is now, Cura will replicate the path from the user's computer to the SD Card.

A preferred method would be to Opt Out of such an option, perhaps via check box?

Reasoning:  In my case, i have my 3D files on a network share, so any computer I use (or even remotely) can access the models.  Due to the long path, when you export to the SD card to print, I am required to go through many clicks to get to the actual file.

It would be better if the gcode was placed on the root of the SD card (as it always had before)

Paul

fieldOfView, 2015-09-25:
I can't really see much of a use-case for the current naming. Perhaps a couple of options could be useful:
* Replicate original path (current default)
* One folder per Year-month-day
* Root
* ...

awhiemstra, 2015-10-01:
Is this still an issue? Both on Windows and Linux it saves to the root of the SD card using the supplied file name.

LilBub, 2015-10-01:
Well, yes.

As someone mentioned, you can remove the path by deleting all the text in the 'Base folder to replicate on SD card' in 15.04.2   

[image]

Otherwise, as I mentioned using a network share at home, you wind up with a very long base folder path.  Which, on the Ultimaker Original Ulticontroller, requires a lot of clicks to finally find your file

I haven't had a chance to install the latest Beta to see how this behaves there.

Again, I think a better way to do this would be through a checkbox option to copy file to root of SD Card or replicate the path from the computer.

awhiemstra, 2015-10-05:
This is an issue relating to Cura 15.04. This repository contains the code for Cura 15.06 and later. In that version, we do not write the directory structure to the SD card and simply always writes to the root of the SD card. For the code for 15.04, see github.com/daid/Cura .

closed 2015-10-05 by awhiemstra