Ultimaker/Cura #433: 15.09.83 - (Crash) need to select printer before selecting global profile

robomaniac, 2015-09-27:
Version: 15.09.83
Platform: Windows-7-6.1.7601-SP1
Qt: 5.4.1
PyQt: 5.4.1

Exception:
Traceback (most recent call last):
  File "c:\dev\uranium\UM\Qt\Bindings\MachineManagerProxy.py", line 73, in setActiveProfile
  File "c:\dev\uranium\UM\Settings\MachineManager.py", line 214, in setActiveProfile
  File "c:\dev\uranium\UM\Signal.py", line 102, in emit
  File "c:\dev\uranium\UM\Qt\Bindings\SettingCategoriesModel.py", line 61, in _onActiveProfileChanged
AttributeError: 'NoneType' object has no attribute 'getMachineDefinition'

----------------------------------------------------
euh... I just open a .STL file, then when in Advanced, Global profile to select the one I wanted then it crash. The printer was not selected, it was just blank background. 

Well, it make sense if you look at the error  "object has no attribute 'getMachineDefinition'

robomaniac, 2015-09-27:
OK, I just downloaded .84 and the very first thing is ask me for my printer. So I can no longer have a blank background. 

I guest it's resolved.

Ghostkeeper, 2015-09-27:
It's one of the things that was recently fixed. Thanks for reporting though!

closed 2015-09-27 by Ghostkeeper