Ultimaker/Cura #435: 15.09.84: Can't write to cura.log / Warning of error not logged

CCS86, 2015-09-27:
Steps to reproduce:

Running Windows 10
Open cura
Create a new printer
Exit Cura

Error:
[image]

Running Cura as administrator avoids this, but gives another error on exit:

[image]

The Cura log shows no actual error:

2015-09-27 09:38:52,808 - INFO - Loaded plugin ConsoleLogger
2015-09-27 09:38:52,868 - INFO - Loaded plugin CuraEngineBackend
2015-09-27 09:38:52,876 - INFO - Loaded plugin OBJReader
2015-09-27 09:38:52,882 - INFO - Loaded plugin OBJWriter
2015-09-27 09:38:52,886 - INFO - Loaded plugin STLReader
2015-09-27 09:38:52,894 - INFO - Loaded plugin STLWriter
2015-09-27 09:38:52,898 - INFO - Loaded plugin FileLogger
2015-09-27 09:38:52,906 - INFO - Loaded plugin GCodeWriter
2015-09-27 09:38:52,914 - INFO - Loaded plugin LayerView
2015-09-27 09:38:52,926 - INFO - Loaded plugin LocalFileOutputDevice
2015-09-27 09:38:52,934 - INFO - Loaded plugin PostProcessingPlugin
2015-09-27 09:38:52,950 - INFO - Loaded plugin RemovableDriveOutputDevice
2015-09-27 09:38:52,970 - INFO - Loaded plugin SliceInfoPlugin
2015-09-27 09:38:52,982 - INFO - Loaded plugin CameraTool
2015-09-27 09:38:52,994 - INFO - Loaded plugin MirrorTool
2015-09-27 09:38:53,010 - INFO - Loaded plugin PerObjectSettingsTool
2015-09-27 09:38:53,030 - INFO - Loaded plugin RotateTool
2015-09-27 09:38:53,046 - INFO - Loaded plugin ScaleTool
2015-09-27 09:38:53,062 - INFO - Loaded plugin SelectionTool
2015-09-27 09:38:53,078 - INFO - Loaded plugin TranslateTool
2015-09-27 09:38:53,095 - INFO - Checking for new version of cura
2015-09-27 09:38:53,095 - INFO - Loaded plugin UpdateChecker
2015-09-27 09:38:53,127 - INFO - Loaded plugin USBPrinting
2015-09-27 09:38:53,158 - INFO - Loaded plugin MeshView
2015-09-27 09:38:53,174 - INFO - Loaded plugin WireframeView
2015-09-27 09:38:53,254 - INFO - Started engine process: C:\Program Files\Cura 15.09\CuraEngine.exe
2015-09-27 09:38:53,363 - DEBUG - Setting value of setting wall_line_count to 5
2015-09-27 09:38:53,363 - DEBUG - Setting value of setting cool_fan_full_layer to 2
2015-09-27 09:38:53,363 - DEBUG - Setting value of setting retraction_prime_speed to 12
2015-09-27 09:38:53,363 - DEBUG - Setting value of setting speed_print to 40
2015-09-27 09:38:53,363 - DEBUG - Setting value of setting skin_outline_count to 1
2015-09-27 09:38:53,364 - DEBUG - Setting value of setting cool_lift_head to True
2015-09-27 09:38:53,364 - DEBUG - Setting value of setting travel_compensate_overlapping_walls_enabled to True
2015-09-27 09:38:53,364 - DEBUG - Setting value of setting adhesion_type to brim
2015-09-27 09:38:53,364 - DEBUG - Setting value of setting retraction_amount to 4
2015-09-27 09:38:53,364 - DEBUG - Setting value of setting retraction_count_max to 3
2015-09-27 09:38:53,364 - DEBUG - Setting value of setting layer_height_0 to 0.2
2015-09-27 09:38:53,365 - DEBUG - Setting value of setting layer_height to 0.2
2015-09-27 09:38:53,365 - DEBUG - Setting value of setting speed_travel to 190
2015-09-27 09:38:53,365 - DEBUG - Setting value of setting remove_overlapping_walls_enabled to True
2015-09-27 09:38:53,365 - DEBUG - Setting value of setting retraction_hop to .1
2015-09-27 09:38:53,365 - DEBUG - Setting value of setting z_seam_type to back
2015-09-27 09:38:53,365 - DEBUG - Setting value of setting support_enable to False
2015-09-27 09:38:53,366 - DEBUG - Setting value of setting z_seam_type to back
2015-09-27 09:38:53,367 - DEBUG - Setting value of setting speed_travel to 190
2015-09-27 09:38:53,367 - DEBUG - Setting value of setting adhesion_type to brim
2015-09-27 09:38:53,367 - DEBUG - Setting value of setting retraction_prime_speed to 12
2015-09-27 09:38:53,367 - DEBUG - Setting value of setting shell_thickness to 1
2015-09-27 09:38:53,367 - DEBUG - Setting value of setting wall_line_count to 5
2015-09-27 09:38:53,368 - DEBUG - Setting value of setting infill_sparse_density to 30
2015-09-27 09:38:53,368 - DEBUG - Setting value of setting layer_height_0 to 0.2
2015-09-27 09:38:53,368 - DEBUG - Setting value of setting support_enable to False
2015-09-27 09:38:53,368 - DEBUG - Setting value of setting material_print_temperature to 216
2015-09-27 09:38:53,368 - DEBUG - Setting value of setting cool_fan_full_layer to 3
2015-09-27 09:38:53,369 - DEBUG - Setting value of setting speed_wall_0 to 40
2015-09-27 09:38:53,369 - DEBUG - Setting value of setting cool_lift_head to True
2015-09-27 09:38:53,369 - DEBUG - Setting value of setting retraction_hop to 0.1
2015-09-27 09:38:53,369 - DEBUG - Setting value of setting speed_print to 40
2015-09-27 09:38:53,370 - DEBUG - Setting value of setting speed_wall_x to 40
2015-09-27 09:38:53,370 - DEBUG - Setting value of setting layer_height to 0.2
2015-09-27 09:38:53,370 - DEBUG - Setting value of setting retraction_amount to 3.65
2015-09-27 09:38:53,370 - DEBUG - Setting value of setting layer_height_0 to 0.2
2015-09-27 09:38:53,370 - DEBUG - Setting value of setting cool_fan_full_layer to 3
2015-09-27 09:38:53,370 - DEBUG - Setting value of setting speed_travel to 190
2015-09-27 09:38:53,370 - DEBUG - Setting value of setting travel_compensate_overlapping_walls_enabled to True
2015-09-27 09:38:53,370 - DEBUG - Setting value of setting skin_outline_count to 1
2015-09-27 09:38:53,370 - DEBUG - Setting value of setting z_seam_type to back
2015-09-27 09:38:53,370 - DEBUG - Setting value of setting coasting_enable to False
2015-09-27 09:38:53,370 - DEBUG - Setting value of setting retraction_amount to 4
2015-09-27 09:38:53,370 - DEBUG - Setting value of setting adhesion_type to brim
2015-09-27 09:38:53,370 - DEBUG - Setting value of setting speed_wall_x to 40
2015-09-27 09:38:53,370 - DEBUG - Setting value of setting layer_height to 0.2
2015-09-27 09:38:53,370 - DEBUG - Setting value of setting wall_line_count to 5
2015-09-27 09:38:53,370 - DEBUG - Setting value of setting retraction_prime_speed to 12
2015-09-27 09:38:53,370 - DEBUG - Setting value of setting retraction_count_max to 3
2015-09-27 09:38:53,370 - DEBUG - Setting value of setting cool_lift_head to True
2015-09-27 09:38:53,370 - DEBUG - Setting value of setting remove_overlapping_walls_enabled to True
2015-09-27 09:38:53,374 - DEBUG - Setting value of setting retraction_hop to .1
2015-09-27 09:38:53,374 - DEBUG - Setting value of setting support_enable to False
2015-09-27 09:38:53,374 - DEBUG - Setting value of setting speed_wall_0 to 40
2015-09-27 09:38:53,374 - DEBUG - Setting value of setting layer_height to 0.06
2015-09-27 09:38:53,374 - DEBUG - Setting value of setting layer_height to 0.1
2015-09-27 09:38:53,378 - DEBUG - Backend connected on port 49674
2015-09-27 09:38:53,390 - DEBUG - Setting active view to MeshView
2015-09-27 09:38:53,513 - DEBUG - Calculating normals took 0.0 seconds
2015-09-27 09:38:53,513 - DEBUG - Loaded a mesh with 10452 vertices
2015-09-27 09:38:53,891 - DEBUG - Loaded theme C:\Program Files\Cura 15.09\resources\themes\cura
2015-09-27 09:38:56,429 - DEBUG - Preparing to send slice data to engine.

Ghostkeeper, 2015-09-27:
Related to #133?

But the second error is new to me.

awhiemstra, 2015-09-28:
It is an issue specific to the 64-bit build of Windows. Numpy complains that the 64-bit build is experimental and will always print that to stdout. Py2Exe then tries its silly log magic that keeps failing.

robomaniac, 2015-09-28:
I have the same error and btw, it's the first Cura (15.09.84) install where I had trouble installing. 
I had error msg that I did not take note :) I just skip.
 My anti-virus (Avast) made the install a little bit longer (that is the 2nd time) it wants to analyse the file.
I think that is normal when you don't release official build.

Normally my .STL file have the Cura logo but with the .84 it's the Fix Corrupted Icons.

closed 2016-12-26 by thopiekar