Ultimaker/Cura #440: Mirror tool applied before rotation

Ghostkeeper, 2015-09-28:
Perhaps this is intended, but it seems a bit unintuitive to me so I'm filing it as a bug.

Version: 15.09.84
Platform: Ubuntu 15.04 Unity
Qt version: 5.4.1
PyQt version: 5.4.1

Steps to reproduce:

* Load a model into Cura.
* Rotate it 90 degrees.
* Mirror it in a direction orthogonal to the axis of rotation that you rotated it in.

Expected behaviour:

* The current state of the model is mirrored.

Observed behaviour:

* The model is mirrored, but it is mirrored as if it were mirrored before the rotation was applied.

nallath, 2017-07-27:
Can no longer reproduce. Probably fixed.

closed 2017-07-27 by nallath

Ghostkeeper, 2017-08-09:
Yeah you fixed this with all your refactoring of the matrix transformations.