Ultimaker/Cura #451: 15.09.85 x64: Crash on Machine Create

CCS86, 2015-10-01:
[image]
[image]

awhiemstra, 2015-10-01:
Looks like your UMO+ config file got corrupted. You also seem to be missing the UMO file. As far as I can tell they are included in the installer, maybe try reinstalling?

Ghostkeeper, 2015-10-01:
How could this happen, though? Maybe with the upgrade from .84 to .85?

CCS86, 2015-10-02:
@awhiemstra , you are right about the UMO file. I changed a setting in there to get the heated bed setting working, and forgot a comma, so it skipped that file. The UMO+ file was unmodified though. 

Now that I know where the cura.log moved, I figured out my syntax error, and created a machine from the UMO file successfully.

closed 2015-10-02 by nallath