Ultimaker/Cura #459: 15.10: start_gcode and end_gcode not commented out in Cura profile string

nicanor-romero, 2015-10-05:
In the Cura profile string the start_gcode and end_gcode have their '\n's parsed as new lines, so these lines do not appear as commented and are actually executed by the printer. I think the '\n's should not be parsed as new lines but instead appear literally as \n characters.

I attach the end of a gcode generated with Cura 15.09.85.

nallath, 2015-10-06:
I think that most machines do handle sending only a \n (atleast marlin simply sends back ok). But you are right, it should be parsed differently.
It should be a fairly easy fix. If you feel like it, you might be able to solve it by yourself.

nicanor-romero, 2015-10-06:
Yes sure. I realized this issue is related to the CuraEngine rather than Cura. But still, I found that the Cura Profile String has been commented out 5 days ago ( https://github.com/Ultimaker/CuraEngine/blob/15.10/src/FffGcodeWriter.cpp#L978 ). I'll have a look at how to parse the end of lines correctly and try to make a pull request.

closed 2016-09-17 by thopiekar