Ultimaker/Cura #465: Printing with USB problem with Ultimaker 2 Go

mpbomil, 2015-10-07:
After pressing "print", the nozzle immediately starts printing without heating up, and the bed also doesn't move to its position. It seems like the start gcode never runs. Printing with SD-card works fine.
I've experienced this with any 15.09.** version so far.

peteruithoven, 2015-10-20:
(This is an issue for Ultimaker2's in general.)
Those machine's gcode flavor are set to Ultigcode, which only properly works from an SD card. 
To support USB printing the gcode flavor of the machine has to be changed to reprap and there needs to be start- and end gcode.
Would it be possible to change the flavor "temporarily" from a plugin? Would it be possible to change the start- and end gcode depending on the gcode flavor? Even easier would be that the settings could be automatically changed based on a usb printing flag (this would hide more implementation details and give more freedom). 

Enabling USB printing is required to support wireless printing using products like the Doodle3D WiFi-Box, Octoprint and Astroprint.

Ghostkeeper, 2015-10-21:
Post processing plug-ins are able to change such things, as a temporary solution.

peteruithoven, 2015-10-27:
But this would require a seperate plugin? And even a setting that users need to change manually?
It also doesn't solve the lack of default start and end gcode for the Ultimaker2, we'd have to add the Ultimaker2 reprap flavor start and end code in our plugin... Shouldn't this be included in the default machine settings of Cura? 
Being able to usb print Ultigcode might be an even better longterm solution, but it will take a while for people to update their firmware.

awhiemstra, 2015-10-27:
The proper solution for this is that the printer should somehow do its start-up sequence when trying to print over USB. However, I do not know if and when this will happen.

companje, 2015-10-27:
I think we need to find out 'if and when' this will happen. Because without this people won't be able to use the USBPrinting-plugin or the Doodle3D-plugin on Ultimaker printers without manually changing their settings files.

Ghostkeeper, 2015-10-28:
I don't see that proper solution happening anytime soon. The firmware developers have their hands full at the moment. Apparently it's fairly involved.

peteruithoven, 2015-10-28:
So would it be possible to have something like a `connection type` setting in Cura? This could filled with things like 'usb' or 'card'. 
This then should change the flavor from ultigcode to reprap and use the necessary start and end gcode. 
This might be implemented by adding connection type specfic machine settings? 

The more hacky solution would be to always enforce the reprap flavor and ignoring the start and end gcode when the ultigcode flavor is used.

awhiemstra, 2015-10-28:
I think the best workaround is to check in the USB plugin whether the flavor is UltiGCode and if so, send some initial GCode commands that resemble the start procedure of the UM2.
@companje Do you think that would be acceptable for the Doodle3D plugin? It would involve a bit of code duplication but it would keep the workaround nicely contained in just the plugin.

I'd like to avoid fiddling around with flavor dynamically, since that has more implications for the slice result than just the start/end code.

Ghostkeeper, 2015-10-28:
You would also have to adjust the extrude rate, though.

peteruithoven, 2015-10-28:
Besides that, we'd have the start and end gcode of a specific device in a plugin. 

Side question, are the start and endgcodes ignores if ultigcode flavor is selected? If that's true the start and end gcode can be added to the regular um2 machine settings.

awhiemstra, 2015-10-28:
start/end code are indeed ignored I think, so that would be a good solution. Combined with the work for template replacement in PR #490 we may have a fairly clean solution for it - if the flavor is UltiGCode, send the start and end gcode as processed to the printer.

peteruithoven, 2015-10-28:
You mean if the flavor isn't UltiGCode send the start and end gcode? 
Maybe I don't understand how the start and end gcode would be processed?

awhiemstra, 2015-10-28:
If the flavor is not UltiGCode, start and end codes are already pre/appended by the engine. So no need to do that manually in the USB printing plugin. However, if the flavor is UltiGCode start and end code are ignored and instead several comments are written. In the case of printing from SD those are read and indicate to the printer that it should perform its start sequence. When printing over USB, it just ignores those comments. So we should instead send a start procedure manually since the printer will not perform it itself.

peteruithoven, 2015-10-28:
Ah, so the UM2 start and end gcode for reprap could be added the machine settings?

awhiemstra, 2015-11-03:
> Ah, so the UM2 start and end gcode for reprap could be added the machine settings?
Yes. Note that this issue will not be resolved until after the 15.10/2.0 release.

companje, 2015-11-03:
Ok, I understand and accept that changing the printer firmware to support UltiGCode over USB or a Doodle3D WiFi-Box is too complicated at this time for a variety reasons.

In the meantime we will try to find the best workaround by approaching it from the Cura / plugin side.

For this there are two things we need from Ultimaker in the next (beta) version of Cura:

1. A function we can call to give us the RepRap flavoured gcode (our code should not need to change the config files).

2. The configuration files for UM2, 2go and Extended should contain valid start- and end-code instead of empty strings.

I really hope you can implement these two things before the release since it's very important for the Doodle3D-WiFi-plugin as well as for the USB-Printing plugin to work with Ultimakers.

nallath, 2015-11-04:
You can already change the flavour of the g-code. It's a setting at the moment, which should enable you to access that by code. 

What we would need to change is some logic that when the flavour is not RepRap the start/end g-code need to be ignored.

awhiemstra, 2015-11-04:
I do not see a reason why the flavour should change. As far as I know, the only difference with volumetric RepRap is appending temperature/start/end GCode commands. So we need to update the start and end GCode of the UM2 profiles, which can be done for release. However, the token replacement code will not be part of the next release since it has been feature frozen for a while. So you will also need to send the temperature commands manually.

Note that you can get the start/end gcode from the active machine instance, since it is a machine setting using: 

    Application.getInstance().getMachineManager().getActiveMachineInstance().getMachineSettingValue("machine_start_gcode")

mpbomil, 2015-11-04:
I've compared the ultigcode and reprap flavored gcode (saved to file after loading in an STL, and realized that the ultigcode has way more extrusion, why is this? I've tried testprinting with the ultigcode with my added start and end code and the print resulted in a big blob of plastic caused by overextrusion. What am I missing here?

UltiGCode:
;TYPE:SKIRT
G1 F900.000 X56.328 Y53.053 E0.15647
G1 X56.584 Y53.054 E0.18708
G1 X57.192 Y53.063 E0.25981
G1 X57.701 Y53.087 E0.32075
G1 X57.941 Y53.106 E0.34955
G1 X57.943 Y53.106 E0.34979
G1 X58.711 Y53.232 E0.44287
G1 X61.509 Y53.649 E0.78120
G1 X64.292 Y53.984 E1.11645

RepRap:

;TYPE:SKIRT
G1 F900.000 X56.328 Y53.053 E0.02453
G1 X56.584 Y53.054 E0.02933
G1 X57.192 Y53.063 E0.04073
G1 X57.701 Y53.087 E0.05028
G1 X57.941 Y53.106 E0.05479
G1 X57.943 Y53.106 E0.05483
G1 X58.711 Y53.232 E0.06942
G1 X61.509 Y53.649 E0.12246
G1 X64.292 Y53.984 E0.17501

nallath, 2015-11-04:
ultigcode uses cubic (mm³) as extrusion whereas reprap uses mm of filament.

closed 2016-12-27 by thopiekar