Ultimaker/Cura #470: Forever slicing

BagelOrb, 2015-10-08:
Actual: Cura is stuck on Slicing... No percentage is shown yet

That happens approx. every 7 times I wait for a slicing to end.

Branch master.

Terminal:
2015-10-08 15:38:38,774 - DEBUG - Preparing to send slice data to engine.
2015-10-08 15:38:38,780 - DEBUG - Sending data to engine for slicing.

Engine log:
End sliced object called. sliced objects 1 object count: 1
Total time elapsed  0.28s.
Print time: 1316
Filament: 1770
Connection reset by peer

I.e. the engine log hasn't been reset yet, hasn't started yet, or is already finished.

BagelOrb, 2015-10-09:
This seems to occur every time after I've saved a gcode to file.

nickthetait, 2016-04-26:
Seems fixed as of 2.1 beta

BagelOrb, 2016-04-26:
Indeed. Thanks

closed 2016-04-26 by BagelOrb