Ultimaker/Cura #472: Non-determinism

BagelOrb, 2015-10-09:
While slicing a model twice via the command line with the same options results in the same output, slicing an object with the same settings via the front-end does not result in the same output.

Profile: Normal Quality
Setting Travel > Combing > Avoid Distance = 0.1

[image]

[combing_evasion.stl.txt](https://github.com/Ultimaker/Cura/files/10714/combing_evasion.stl.txt)

BagelOrb, 2015-10-09:
Note that this might still be a CuraEngine issue, since the engine is not restarted after each slicing.

BagelOrb, 2015-10-09:
Every time after the terminal says "DEBUG - Killing engine process" 
The resulting gcode looks like:
[image]

BagelOrb, 2015-10-09:
@awhiemstra How much work did you spend on getting the engine to keep running instead of restarting it every slicing?

BagelOrb, 2015-10-09:
Just verified that it's not a visualization bug.

BagelOrb, 2015-10-21:
For the time being this should be solved by the frontend, by restarting the engine on each slice.

TMHogenhout, 2015-10-21:
I have encountered the same issue. I noticed it when I tried to test the outcomes of the postprocessing plugin. Results were not as expected so I started to count the different lines between two slices of the same model. Here's what I did: 
load a cube -> slice it -> save it -> reslice it -> save it -> compare files -> count differences (or better yet make the comp do it)

Here are some outcomes: 
[simplecube with UMO as machine] v.s. [simplecube with UMO as machine] → 1243 lines different
[simplecube with UM2 as machine] v.s. [simplecube with UM2 as machine] → 2262 lines different
When you perform the same steps several times, the outcomes are different as well, but still in the same ballpark. 

Here are some outcomes of some tests I did to test the postprocessing scripts (PAH = Pause At Height; TAZ = TweatAtZ) (all outcomes are with the UM2 and the standard settings):
[simplecube with PAH-script] v.s [simplecube where PAH-script is deleted] → 2272 lines different
[simplecube] v.s. [simplecube where PAH-script is deleted] → 2256 lines different

And then somehow the problem seems lessened when you only add a script and reslice: 
[simplecube] v.s. [simplecube with PAH-script] → 340 lines different 
[simplecube] v.s. [simplecube with TAZ-script] → 330 lines different 

The Engine-log doesn't show any errors. But the Cura log keeps showing this one everytime it slices:
2015-10-21 12:27:38,682 - DEBUG - Sending data to engine for slicing.
Exception occured <urlopen error [Errno 111] Connection refused>

@BagelOrb Please keep me posted on how this issue evolves, because it makes other testing a pain in the neck.

TMHogenhout, 2015-10-21:
The problem was in a way solved by making the engine restart even though it still said 'connection refused'. The two files now counted 0 differences. 

```2015-10-21 14:04:40,816 - DEBUG - Setting value of setting layer_height to 0.1
2015-10-21 14:04:41,350 - DEBUG - Killing engine process
2015-10-21 14:04:42,180 - INFO - Backend crashed or closed. Restarting...
2015-10-21 14:04:42,448 - INFO - Started engine process: /home/tamara/dev/bin/CuraEngine
2015-10-21 14:04:42,454 - DEBUG - Backend connected on port 49674
2015-10-21 14:04:42,934 - DEBUG - Preparing to send slice data to engine.
2015-10-21 14:04:43,053 - DEBUG - Sending data to engine for slicing.
Exception occured <urlopen error [Errno 111] Connection refused>```

BagelOrb, 2015-10-22:
That seems like an independent bug, which you should make a new issue for.

Do you get this bug every time you restart the engine?

Also when the engine is killed and restarted again (when changing settings during slicing)?

Do you kill the engine before restarting it?

TMHogenhout, 2015-10-22:
yeah the exception is probably a different issue, but the other results are not. If I kill the engine if does give the same results though.

BagelOrb, 2017-07-21:
issue is too old

closed 2017-07-21 by BagelOrb

Ghostkeeper, 2017-07-27:
Also haven't seen this in a very long time. I believe we fixed it.