Ultimaker/Cura #477: Layer Slider Deadzone / Sensitivity

CCS86, 2015-10-12:
I have been struggling with the layer slider, especially on prints with higher layer counts. There seems to be an excessively large deadzone, followed by overly sensitive movement. It is nearly impossible to move only one layer, and it seems that the layer keyboard shortcut is no longer functional.

nallath, 2015-10-13:
The keyboard shortcut is now ctrl + arrow key.

CCS86, 2015-10-14:
I'll try that (could have sworn I tried every modifier key + arrow combination)

HAve you noticed the deadzone and sensitivity feeling off on large prints?

nallath, 2015-10-15:
Not really, but to be honest i've only tried it with smaller prints.

CCS86, 2015-10-15:
I noticed that one of the more intrusive aspects is actually that there is a big lag in the layer control. It seems to wait until the drawing of the layers is complete before updating the slider position. 

It seems much better if instead, the slider was independent of layer drawing. Meaning that you could have immediate control of layer slider position, then the layer drawing would follow.

nallath, 2015-10-15:
It should be independent of drawing, but python has a the infamous GIL which messes with things.

BagelOrb, 2017-07-21:
not a problem any more

closed 2017-07-21 by BagelOrb