Ultimaker/Cura #482: Import a model and need to rotate

gpb01, 2015-10-15:
When you import a model from many popular modelling programs you have to rotate the model. Changing behavior could create difficulties for current users who are used to this behavior, but should be possible to create an On/Off switch, in Cura General Preference, to given the choice to the user if rotate or not on import.

Ghostkeeper, 2015-10-16:
You mean, because the modelling program uses a different coordinate system where Z is not the vertical direction (against gravity) but the direction towards the back of the printer?

It'd be nice if we could detect that automatically in some way. It's not reliable perhaps, but many of these programs leave a comment in the files they generate that says that it was generated with their application...

awhiemstra, 2015-10-16:
> It'd be nice if we could detect that automatically in some way. It's not reliable perhaps, but many of 
> these programs leave a comment in the files they generate that says that it was generated with their 
> application...

That would mean maintaining a database of modelling programs, versions, and coordinate systems. That would be way too much work. What I would rather do is create an import dialog when loading files where you can specify these settings, optionally storing them and hiding the import dialog.

gpb01, 2015-10-18:
Sorry for the late answer ... I was out of office ...
... just to clarify, my enhancement request originates from this discussion : https://ultimaker.com/en/community/view/17296-cura-1510-open-beta?page=5&sort=#reply-121487

BagelOrb, 2017-07-21:
Duplicate of https://github.com/Ultimaker/Cura/issues/71

closed 2017-07-21 by BagelOrb