Ultimaker/Cura #484: Calculating and slicing phase button

warmwhisky, 2015-10-16:
When you make a setting change like rotation, scaling, layer height, enabling support etc etc, the slicing/calculating restarts. I if need to rotate it twice, then scale it and what ever other changes, it wastes both my CPU and time. On top of this I am finding the more setting changes I make the more likely it is to crash the software. Then I have to go through the painful process again, trying all manor of different ways to stop it crashing.

A simple button to start calculation/slicing phase, after I have made my adjustments is going to bring my crash rate and frustration levels down to a minimum.

Ghostkeeper, 2015-10-16:
Duplicate of #12 and #109.

Disabling auto-slicing is something that has been requested often by the community but the current stance is "no, since auto-slicing shouldn't affect the front-end in any way". There are still some bugs left where the slicing does slow down the front-end (in particular, the communication with the slicing engine). It might be possible to write a plug-in for it though.

warmwhisky, 2015-10-16:
So auto-slicing "shouldn't" affect the front-end in any way, but it does.

all the way back to issues #12 and still its here.

I cant use the software because of this. I'll close it as it a plain No.

closed 2015-10-16 by warmwhisky

nallath, 2015-10-16:
We want to fix the issue, but we won't apply a band aid patch to remove the symptoms of the problem. If you can tell us what causes the slicing to crash, we can fix this. 

It's not the auto slicing that's the problem; the problem is that some setting combinations that you managed to find crash the software. If we would disable auto slicing, you would still have this problem, but at a later stage of this problem.

warmwhisky, 2015-10-16:
Im using 15.09.90.

Its happening when I rotate. My model is lying on its back. I need to simply rotate it once 90 degrees (everytime) but the new rotate tool is far from stable - #485 so I have to rotate it 2 or 3 times before I fluke it into place.

warmwhisky, 2015-10-16:
I just made the rotation in 3D Max to fit with Cura so I did not need any rotation. I successfully imported the largest OBJ I have imported; 230mb and now it is printing. Calculation phase is a painful thing for me.

Surely it is more complex to code in a clause to start slicing on every function?