Ultimaker/Cura #496: Switching between Simple and Advanced Settings is slow

Xisle, 2015-10-20:
Cura 15.09.91
OS X 10.9.5

My guess is the settings panel options are rebuilt every time a user switches between Simple and Advanced settings. There is a noticeable delay when switching to Simple and a 3-4 second delay when switching back to Advanced.

The Advanced options are also displayed uncollapsed then quickly collapsed to match the user's previous settings.

daid, 2015-12-17:
Did some tracing on this. Found that the "setActiveProfile" on the MachineManager is one of (but most likely not the only) slow functions in this case.

I changed the signal emit of this function to the following:
```
        import profile
        profile.run("from UM.Application import Application\nApplication.getInstance().getMachineManager().activeProfileChanged.emit()")
```
Which produced the following trace output:
```
         424029 function calls (319419 primitive calls) in 3.260 seconds

   Ordered by: standard name

   ncalls  tottime  percall  cumtime  percall filename:lineno(function)
        8    0.000    0.000    0.000    0.000 :0(abs)
     2273    0.000    0.000    0.000    0.000 :0(add)
      967    0.000    0.000    0.000    0.000 :0(append)
      128    0.000    0.000    0.000    0.000 :0(array)
      436    0.000    0.000    0.000    0.000 :0(beginInsertRows)
       22    0.000    0.000    0.000    0.000 :0(beginResetModel)
        6    0.000    0.000    0.000    0.000 :0(ceil)
       30    0.016    0.001    0.016    0.001 :0(clear)
     3088    0.000    0.000    0.000    0.000 :0(decode)
       42    0.000    0.000    0.000    0.000 :0(emit)
       48    0.000    0.000    0.000    0.000 :0(encode)
      436    1.638    0.004    1.669    0.004 :0(endInsertRows)
       22    0.031    0.001    0.031    0.001 :0(endResetModel)
      748    0.000    0.000    0.000    0.000 :0(eval)
        1    0.000    0.000    3.260    3.260 :0(exec)
      740    0.000    0.000    0.000    0.000 :0(extend)
4401/4397    0.016    0.000    0.016    0.000 :0(format)
      163    0.000    0.000    0.000    0.000 :0(get)
        4    0.000    0.000    0.000    0.000 :0(get_ident)
      748    0.000    0.000    0.000    0.000 :0(globals)
        1    0.000    0.000    0.000    0.000 :0(hasattr)
       18    0.000    0.000    0.000    0.000 :0(id)
       80    0.000    0.000    0.000    0.000 :0(index)
      436    0.000    0.000    0.000    0.000 :0(insert)
     3088    0.000    0.000    0.000    0.000 :0(isValid)
     1946    0.000    0.000    0.000    0.000 :0(isinstance)
      576    0.000    0.000    0.000    0.000 :0(items)
        3    0.000    0.000    0.000    0.000 :0(iter)
        4    0.000    0.000    0.000    0.000 :0(keys)
     5772    0.000    0.000    0.000    0.000 :0(len)
      170    0.000    0.000    0.000    0.000 :0(max)
      160    0.000    0.000    0.000    0.000 :0(min)
       16    0.000    0.000    0.000    0.000 :0(reduce)
      445    0.000    0.000    0.000    0.000 :0(remove)
     2192    0.000    0.000    0.000    0.000 :0(replace)
        2    0.000    0.000    0.000    0.000 :0(repr)
       17    0.000    0.000    0.000    0.000 :0(resize)
        2    0.000    0.000    0.000    0.000 :0(round)
     3088    0.000    0.000    0.000    0.000 :0(row)
        1    0.000    0.000    0.000    0.000 :0(setprofile)
      128    0.000    0.000    0.000    0.000 :0(sleep)
        4    0.000    0.000    0.000    0.000 :0(sort)
        2    0.000    0.000    0.000    0.000 :0(start)
        1    0.000    0.000    0.000    0.000 :0(update)
       18    0.000    0.000    0.000    0.000 :0(zeros)
        1    0.000    0.000    0.000    0.000 <frozen importlib._bootstrap>:2264(_handle_fromlist)
        1    0.000    0.000    3.260    3.260 <string>:1(<module>)
        4    0.000    0.000    0.000    0.000 Application.py:217(getMachineManager)
        2    0.000    0.000    0.000    0.000 Application.py:232(getController)
        4    0.000    0.000    0.000    0.000 Application.py:265(getMainThread)
        5    0.000    0.000    0.000    0.000 Application.py:269(getInstance)
        3    0.000    0.000    0.000    0.000 AxisAlignedBox.py:16(__init__)
        3    0.000    0.000    0.000    0.000 AxisAlignedBox.py:218(_ensureMinMax)
        2    0.000    0.000    0.000    0.000 AxisAlignedBox.py:234(__repr__)
        1    0.000    0.000    0.031    0.031 BuildVolume.py:168(_onActiveProfileChanged)
        1    0.016    0.016    0.016    0.016 BuildVolume.py:183(_updateDisallowedAreas)
[... 274 lines cut ...]
        2    0.000    0.000    0.000    0.000 wireframe_drag_along:1(<module>)
        2    0.000    0.000    0.000    0.000 wireframe_fall_down:1(<module>)
        2    0.000    0.000    0.000    0.000 wireframe_flat_delay:1(<module>)
        6    0.000    0.000    0.000    0.000 wireframe_flow:1(<module>)
        4    0.000    0.000    0.000    0.000 wireframe_flow_connection:1(<module>)
        4    0.000    0.000    0.000    0.000 wireframe_flow_flat:1(<module>)
        2    0.000    0.000    0.000    0.000 wireframe_height:1(<module>)
        2    0.000    0.000    0.000    0.000 wireframe_nozzle_clearance:1(<module>)
        2    0.000    0.000    0.000    0.000 wireframe_printspeed:1(<module>)
        4    0.000    0.000    0.000    0.000 wireframe_printspeed_bottom:1(<module>)
        4    0.000    0.000    0.000    0.000 wireframe_printspeed_down:1(<module>)
        4    0.000    0.000    0.000    0.000 wireframe_printspeed_flat:1(<module>)
        4    0.000    0.000    0.000    0.000 wireframe_printspeed_up:1(<module>)
        2    0.000    0.000    0.000    0.000 wireframe_roof_drag_along:1(<module>)
        2    0.000    0.000    0.000    0.000 wireframe_roof_fall_down:1(<module>)
        4    0.000    0.000    0.000    0.000 wireframe_roof_inset:1(<module>)
        2    0.000    0.000    0.000    0.000 wireframe_roof_outer_delay:1(<module>)
        6    0.000    0.000    0.000    0.000 wireframe_straight_before_down:1(<module>)
        2    0.000    0.000    0.000    0.000 wireframe_strategy:1(<module>)
        2    0.000    0.000    0.000    0.000 wireframe_top_delay:1(<module>)
        2    0.000    0.000    0.000    0.000 wireframe_top_jump:1(<module>)
        2    0.000    0.000    0.000    0.000 wireframe_up_half_speed:1(<module>)
```

Surprising (to me), there are 22 calls to the: SettingsFromCategoryModel.py:176(_onProfileChanged) function.

nallath, 2016-01-13:
I've also ran the test but on linux (from source). I got the following data
```
perty change signal or superclass signal
         551473 function calls (417980 primitive calls) in 2.189 seconds

   Ordered by: standard name

   ncalls  tottime  percall  cumtime  percall filename:lineno(function)
        1    0.000    0.000    0.000    0.000 :0(abs)
     3714    0.003    0.000    0.003    0.000 :0(add)
      902    0.001    0.000    0.001    0.000 :0(append)
       60    0.000    0.000    0.000    0.000 :0(array)
      436    0.004    0.000    0.005    0.000 :0(beginInsertRows)
       24    0.000    0.000    0.000    0.000 :0(beginResetModel)
        9    0.000    0.000    0.000    0.000 :0(ceil)
       32    0.001    0.000    0.001    0.000 :0(clear)
       99    0.001    0.000    0.001    0.000 :0(compile)
     3125    0.003    0.000    0.003    0.000 :0(decode)
       52    0.004    0.000    0.006    0.000 :0(emit)
       48    0.000    0.000    0.000    0.000 :0(encode)
      436    0.990    0.002    1.043    0.002 :0(endInsertRows)
       24    0.016    0.001    0.016    0.001 :0(endResetModel)
     1122    0.002    0.000    0.003    0.000 :0(eval)
        1    0.000    0.000    2.189    2.189 :0(exec)
      931    0.001    0.000    0.001    0.000 :0(extend)
4410/4406    0.005    0.000    0.005    0.000 :0(format)
       70    0.000    0.000    0.000    0.000 :0(get)
        4    0.000    0.000    0.000    0.000 :0(get_ident)
     1122    0.001    0.000    0.001    0.000 :0(globals)
        1    0.000    0.000    0.000    0.000 :0(hasattr)
       24    0.000    0.000    0.000    0.000 :0(id)
       96    0.000    0.000    0.001    0.000 :0(index)
      436    0.001    0.000    0.001    0.000 :0(insert)
     3125    0.003    0.000    0.003    0.000 :0(isValid)
     4549    0.003    0.000    0.003    0.000 :0(isinstance)
      577    0.001    0.000    0.001    0.000 :0(items)
        3    0.000    0.000    0.000    0.000 :0(iter)
     4503    0.003    0.000    0.003    0.000 :0(len)
       55    0.000    0.000    0.000    0.000 :0(max)
       48    0.000    0.000    0.000    0.000 :0(min)
        6    0.000    0.000    0.000    0.000 :0(reduce)
     1499    0.001    0.000    0.001    0.000 :0(remove)
     2295    0.002    0.000    0.002    0.000 :0(replace)
        2    0.000    0.000    0.000    0.000 :0(repr)
       11    0.000    0.000    0.000    0.000 :0(resize)
        3    0.000    0.000    0.000    0.000 :0(round)
     3125    0.002    0.000    0.002    0.000 :0(row)
        1    0.000    0.000    0.000    0.000 :0(setprofile)
        2    0.000    0.000    0.000    0.000 :0(start)
        1    0.000    0.000    0.000    0.000 :0(update)
       14    0.000    0.000    0.000    0.000 :0(zeros)
        1    0.000    0.000    0.000    0.000 <frozen importlib._bootstrap>:2264(_handle_fromlist)
        1    0.000    0.000    2.189    2.189 <string>:1(<module>)
        1    0.000    0.000    0.059    0.059 ActiveProfileProxy.py:35(_onActiveProfileChanged)
        1    0.000    0.000    0.059    0.059 ActiveProfileProxy.py:46(_onSettingValuesChanged)
        5    0.000    0.000    0.000    0.000 Application.py:219(getMachineManager)
        2    0.000    0.000    0.000    0.000 Application.py:234(getController)
        4    0.000    0.000    0.000    0.000 Application.py:267(getMainThread)
        6    0.000    0.000    0.000    0.000 Application.py:271(getInstance)
        3    0.000    0.000    0.000    0.000 AxisAlignedBox.py:16(__init__)
        3    0.000    0.000    0.000    0.000 AxisAlignedBox.py:218(_ensureMinMax)
        2    0.000    0.000    0.000    0.000 AxisAlignedBox.py:234(__repr__)
        1    0.000    0.000    0.014    0.014 BuildVolume.py:175(_onActiveProfileChanged)
        1    0.000    0.000    0.005    0.005 BuildVolume.py:190(_updateDisallowedAreas)
        2    0.000    0.000    0.007    0.004 BuildVolume.py:253(_getSkirtSize)
       48    0.000    0.000    0.001    0.000 BuildVolume.py:274(_clamp)
        1    0.000    0.000    0.009    0.009 BuildVolume.py:77(rebuild)
        1    0.000    0.000    0.000    0.000 Color.py:10(__init__)
       72    0.000    0.000    0.000    0.000 Color.py:16(r)
       72    0.000    0.000    0.000    0.000 Color.py:23(g)
[... 272 lines cut ...]
        6    0.000    0.000    0.000    0.000 wireframe_drag_along:1(<module>)
        6    0.000    0.000    0.000    0.000 wireframe_fall_down:1(<module>)
        6    0.000    0.000    0.000    0.000 wireframe_flat_delay:1(<module>)
        6    0.000    0.000    0.000    0.000 wireframe_flow:1(<module>)
        9    0.000    0.000    0.000    0.000 wireframe_flow_connection:1(<module>)
        9    0.000    0.000    0.000    0.000 wireframe_flow_flat:1(<module>)
        6    0.000    0.000    0.000    0.000 wireframe_height:1(<module>)
        6    0.000    0.000    0.000    0.000 wireframe_nozzle_clearance:1(<module>)
        6    0.000    0.000    0.000    0.000 wireframe_printspeed:1(<module>)
        9    0.000    0.000    0.000    0.000 wireframe_printspeed_bottom:1(<module>)
        9    0.000    0.000    0.000    0.000 wireframe_printspeed_down:1(<module>)
        9    0.000    0.000    0.000    0.000 wireframe_printspeed_flat:1(<module>)
        9    0.000    0.000    0.000    0.000 wireframe_printspeed_up:1(<module>)
        6    0.000    0.000    0.000    0.000 wireframe_roof_drag_along:1(<module>)
[... cut ...]
        6    0.000    0.000    0.000    0.000 wireframe_straight_before_down:1(<module>)
        2    0.000    0.000    0.000    0.000 wireframe_strategy:1(<module>)
        6    0.000    0.000    0.000    0.000 wireframe_top_delay:1(<module>)
        6    0.000    0.000    0.000    0.000 wireframe_top_jump:1(<module>)
        6    0.000    0.000    0.000    0.000 wireframe_up_half_speed:1(<module>)
        4    0.000    0.000    0.000    0.000 xy_offset:1(<module>)```

It seems that Linux handles certain things significantly faster ( ListModel.py:56(appendItem) (1.6 sec vs 1.0)

daid, 2016-01-13:
Could also be you have a faster machine.
But linux is a lot faster with memory allocation.

Time spend on the "SettingsCategory.py:89(getSetting)" that we talked about still holds up in your trace as well.

Ghostkeeper, 2016-01-14:
300.000 calls to `Setting::getSetting`?

And `endInsertRows` is apparently very slow. Maybe we can find an alternative for that.

fieldOfView, 2017-06-12:
This has been fixed; issue can be closed

closed 2017-06-13 by awhiemstra