Ultimaker/Cura #497: UMO platform image turned black after loading .stl

Xisle, 2015-10-20:
Cura 15.09.91
OS X 10.9.5

I loaded an .stl and got this..

[image]

Ghostkeeper, 2015-10-21:
Looks like the normals got erased / corrupt.

Ghostkeeper, 2015-10-21:
I can't find the results of @TMHogenhout's OSX test cases, but in the others this error hasn't occurred. It will be difficult to reproduce this.

TMHogenhout, 2015-10-21:
Well actually I have seen this many times, in another form though. I didn't put it in the test report because it wasn't within the scope of the test (which was only targeted at the stability of Cura). It doesn't happen on my computer (testing from source), but I think I can reproduce it on one of the test-computers (to be continued..)

fieldOfView, 2016-11-09:
Issue can be closed. Was fixed by making sure there is always a texture, even if the machine definition does not specify one.

closed 2016-11-09 by nallath