Ultimaker/Cura #500: Loading G-code separately

Ghostkeeper, 2015-10-22:
At times it is useful to load in G-code separately, without having a model. Cura 15.04 had this feature but it hasn't been implemented yet after the rework, so this was a regression. I'm making this issue to put it up as a possibility for Cura's roadmap.

There are several reasons to load G-code separately, that I can think of:

* Slicing with a back-end that hasn't yet been integrated in a plug-in for Cura.
* When a service only provides G-code to download, rather than the original model.
* Verifying that the written G-code file is actually the correct output.
* Looking at which files are on a printer's SD card.

If accepted, this feature would have to keep in mind how to handle selection and having multiple objects on the platform. In Cura 15.04, pure G-code objects could not be selected and they were only visible in the layer view. I think it makes some sense to not allow combining pure G-code objects with normal models: Loading a G-code file erases any other objects from the platform, and loading any normal mesh erases any pure G-code from the platform.

I'd like to hear your thoughts!

awhiemstra, 2015-10-22:
Yes, it is something that we actually did some initial work on in the master branch. One of the reasons it is not there yet is that I did not want to make a main feature of Cura (the layer view) dependant on parsing GCode.

One thing I dislike about how it is implemented in 15.04 is that there is no indication about the fact that you have GCode loaded. We just did not find a proper solution for that yet.

nickthetait, 2016-04-18:
Is this feature targeted for the 2.2 release?

nallath, 2016-04-19:
I'm not a 100% sure, but i think it's 2.3 right now.

Ghostkeeper, 2016-04-19:
Nop, 3.0: CURA-1063.

nallath, 2016-04-19:
Really? Wow. That suprises me. I think we should move this to an earlier release.

BagelOrb, 2017-07-21:
implemented

closed 2017-07-21 by BagelOrb