Ultimaker/Cura #503: Ubuntu 15.04 - Cura crashes with a segfault

danilius, 2015-10-22:
I have Ubuntu 15.04 installed, and Cura crashes with a segfault, although the splashscreen does show up for a moment, as well as the changelog window, and then everything disappears

I have python3-opengl installed, but that does not seem to be the problem.

I do have Nvidia drivers installed, version 346.96.

Output from the command line can be found here: http://daniepstein.com/daniepstein/wp-content/uploads/2015/10/cura-crash-report.txt

daid, 2015-10-23:
This should be fixed in the latest beta:
http://software.ultimaker.com/Cura_closed_beta/15.10/?C=M;O=D
(odd conflict between Mesa development packages and the driver on Ubuntu)

danilius, 2015-10-23:
Sadly no.

awhiemstra, 2015-10-23:
"QOpenGLShader: could not create shader" that does suggest your 3D support fails to load properly or is broken. Can you provide the output of glxinfo ?

danilius, 2015-10-23:
You can find it here: http://daniepstein.com/daniepstein/wp-content/uploads/2015/10/glxinfo_output.txt

danilius, 2015-11-01:
Has there been any progress on this issue? I have researched this myself but go nowhere. It's not really my field.

Ghostkeeper, 2015-11-02:
We haven't reproduced this yet. Not on Ubuntu with Intel drivers at least. We have a testing computer with NVidia and Ubuntu, it could be tried there.

TMHogenhout, 2015-11-02:
@danilius  Which version of Cura are your using?

danilius, 2015-11-02:
15.09.91

danilius, 2015-12-11:
Solved it. I modified the beginning of cura_app.py to read as follows:

#!/usr/bin/env python3

# Copyright (c) 2015 Ultimaker B.V.
# Cura is released under the terms of the AGPLv3 or higher.

import sys
from OpenGL import GL

Ghostkeeper, 2015-12-14:
Oh, good job! I wonder though why cura_app.py would need OpenGL though, since there is no direct reference to it there. Perhaps the problem is better solved when importing OpenGL in different files.

daid, 2015-12-14:
https://github.com/Ultimaker/Cura/blob/master/cura/CuraApplication.py#L57

danilius, 2015-12-14:
I will give that a whirl when I have a moment and see if that works.

thopiekar, 2016-01-14:
@Ghostkeeper: I also think it should be moved into Uranium.
I remember I also reported about this segfault already, but closed it as I thought it is KDE related, so as I was working with my notebook with Intel graphics I was sure it is fixed now.

However, as far as I understand PyQt5 as a Qt5 binding, it shouldn't depend on Python-OpenGL. So what I believe is that Python-OpenGL has something in __init__ that prepares the hardware to work, what Qt5 doesn't seem to do. So the only true solution to this problem is to remove this dirty fix and fix it the PyQt5 way.

closed 2016-09-19 by thopiekar