Ultimaker/Cura #504: Layerview: different extruder => different nozzle

BagelOrb, 2015-10-23:
Make it so that different extruder trains are visualized differently.

Make sendPolygons of CuraEngine also send extruder_nr.

Ghostkeeper, 2015-10-26:
Requires some thought on how to differentiate the colours between multiple nozzles, while keeping the associations of colours with walls, support, etc. consistent. Perhaps we should vary the saturation? Pastel vs. deep colours?

For one thing, I think it'd be good to select two colours and vary linearly between them, since that will generalise well to more than two nozzles.

BagelOrb, 2015-10-26:
That or make it togglable for what you use different colors: either visualize as it is now, or color the segments based solely on extruder_nr and switch between these two with a button.

fieldOfView, 2016-11-09:
Layers are now shown in a darker color if they are printed with an "inactive" extruder. The "active" extruder is determined by the currently selected extruder-tab.

Issue can be closed

closed 2016-11-09 by nallath