Ultimaker/Cura #505: wrong %userprofile% folder on itinerant profile (W7)

ljoets, 2015-10-26:
Hello,
I have to deploy latest stable Cura ( version 15.04.2) on W7x64 computers in a school.
Users have itinerant windows profiles on a sambaedu3 server.
In theory, %userprofile% is on a network drive. And I expected to find .cura folder on it, but .cura folder stays on C:\Users\<login>\
Is there a method to specify %userprofile% at launch ?
Thanks.

Ghostkeeper, 2015-10-27:
Not easily. In util/profile.py it says on lines [645](https://github.com/daid/Cura/blame/SteamEngine/Cura/util/profile.py#L645) and 646:

    if platform.system() == "Windows":
        basePath = os.path.normpath(os.path.expanduser('~/.cura/%s' % version.getVersion(False)))

So it saves the settings always in <user profile>/.cura

You could change the source code in that line and re-package, but that version of Cura doesn't yet support changing the settings directory.

ljoets, 2015-10-27:
Ok, thanks.
~/.cura/%s seems to use C:\Users\<login>\.cura\ directory.
Do you know what I have to use in place of ~ to go in %userprofile% (windows var that points to itinerant profile if its exists) ?

awhiemstra, 2015-10-27:
I would expect os.path.expanduser() to use %userprofile% for the home directory if that is a standard variable to indicate that. I am afraid you will need to look at python's source code if you want to know exactly how it works.

If you want a quick hack, using os.path.join(os.environ["userprofile"], ".cura") would get you the right directory instead of os.path.expanduser().

ljoets, 2015-10-27:
os.path.join(os.environ["userprofile"], ".cura") does not use itinerant profile, but still C:\Users\<login>\ folder.
That quick & dirty trick works for me... Specify K:\profil\ network drive in hard...
In profile.py file, line 646
`#basePath = os.path.normpath(os.path.expanduser('~/.cura/%s' % version.getVersion(False)))
basePath = os.path.normpath(os.path.expanduser('K:/profil/.cura/%s' % version.getVersion(False)))`
So I deploy new profile.py file after silent install via wpkg.

Thanks for help !

Ghostkeeper, 2015-10-28:
Glad it works!

closed 2015-10-28 by Ghostkeeper