Ultimaker/Cura #506: Support generated mid air

ETitus, 2015-10-26:
Application Version: 15.09.91

Platform: Windows 10

Qt: ????

PyQt: ??????

Display Driver: 
Driver Name: nvd3dumx.dll,nvwgf2umx.dll,nvwgf2umx.dll,nvwgf2umx.dll,nvd3dum,nvwgf2um,nvwgf2um,nvwgf2um
Driver File Version: 10.18.0013.5850 (English)
     Driver Version: 10.18.13.5850

Steps to Reproduce:
Load the model and select standard support: http://www.thingiverse.com/download:1520868

Actual Results:
Support is generated in mid air for some parts of the support

Expected results:
Like previous versions, generate support always from the model itself

Additional Information:
The support starts mid air! What the hell? The goal of support is to not start in mid air xD
It looks like it takes the highest layer it has to put support on to start on, rather than let it grow. I checked 15.04.2 and that worked correctly. 

Screenshots:
https://d2py9w124w2itd.cloudfront.net/photo/image/2500x0/562a43fad9194.PNG
https://d2py9w124w2itd.cloudfront.net/photo/image/2500x0/562a440576eec.PNG
Also see: https://ultimaker.com/en/community/view/17296-cura-1510-open-beta?page=5&sort=#reply-122355

Xisle, 2015-10-26:
I get the same on the lower supports. I can't quite tell if the ones around the center piece look ok.

[image]

awhiemstra, 2015-10-26:
@BagelOrb this looks like an engine issue.

BagelOrb, 2015-11-04:
The bottom distance of the support is violated by the Stair Step Height setting. Set it to 0.0 and support always sits on the model.

However, a stair step height of 0 and ZigZag support infill cause disconnected support lines right on the bottom of the support. 

I'm trying to figure out a solution, but it's going to be hard.

BagelOrb, 2015-11-04:
For now I'm setting the stair step height by default to 0.3mm and the support bottom distance to 0.1mm.

ETitus, 2015-11-04:
Well the old versions of cura do the job correctly, so perhaps something went wrong during the changes from legacy to new?

BagelOrb, 2015-11-04:
The old Cura didn't have ZigZag support.

ZigZag support can support areas smaller than the distance between the support lines, so it has certain advantages.

BagelOrb, 2015-11-04:
Stair Step Height was introduced in the new Cura to help with some of the disadvantages of ZigZag support.

BagelOrb, 2015-11-16:
Fixed in new release: the default distance between the model and the bottom of the support is between 0.1 and 0.3 mm.

closed 2015-11-16 by BagelOrb

ETitus, 2015-12-04:
Hm, still kinda happening? This is the Version: 15.04.03 Release date: 11/4/15
Was it fixed in that one too?

http://i.imgur.com/WJGxkxl.png

BagelOrb, 2015-12-04:
Nope. It's fixed in version 2.1: the new Cura

ETitus, 2015-12-04:
Has that one been released already?

BagelOrb, 2015-12-05:
There have been beta releases of Cura 2, but there hasn't been an official
one yet.