Ultimaker/Cura #510: help nedded: ImportError: "No Module named UM"

smsunny, 2015-10-27:
Hello, I am a newbie programmer in python. I was trying to run the 'Cura' project in Eclipse. I have cloned the github repository in Eclipse and created a PyDev project. But whenever I am trying to run any .py file, it creates error: "ImportError: No module named 'UM'". I am not sure what to do now. Can anyone please help me how can I fix this? Thanx in Advance

Xisle, 2015-10-28:
Grab Uranium also.. not certain what else you need.

https://github.com/Ultimaker/Uranium

smsunny, 2015-10-28:
I have cloned uranium also, but as a separate project. Do I need to have all the files of Uranium in the same directory as Cura?

awhiemstra, 2015-10-28:
No, you need to set the right PYTHONPATH enironment variable. Not sure how Eclipse handles this, but from the command line you would do PYTHONPATH=/path/to/uranium/repo cura_app.py .

closed 2015-10-28 by nallath

zhenhuashao, 2022-03-18:
请问您这边解决了嘛

zhenhuashao, 2022-03-18:
my prompt is wrong：ModuleNotFoundError: No module named 'UM'

Ghostkeeper, 2022-04-10:
On Windows 10 it works the same way. You need to change the `PYTHONPATH` environment variable to include the path to Uranium. I think [this search result](https://superuser.com/a/949577) explains it well enough. It's the same as any other Windows release, really.